ArduPilot/pymavlink #2: DFReader broken for PX4

devbharat, 2016-09-16:
Uses function mode_string_px4 which is no longer there.

peterbarker, 2016-09-16:
Please provide a failing log.

devbharat, 2016-09-16:
I use MAVExplorer which uses DFReader which calls the function `mode_string_px4(mode_number)` which was removed in [this commit](https://github.com/ArduPilot/pymavlink/commit/e67f707c4c56b010c962300516c83f6df3e54a8c), but not replaced in the DFReader.py file. 

Pretty much any PX4 log will fail when read with DFReader( as done by MAVExplorer). For example, this
http://logs.uaventure.com/view/rV9PLoSWvQHhd2JzTuuoFJ.

peterbarker, 2016-09-16:
I've pushed in a fix for this.  The log you supplied now seems to parse and graph OK.

Closing for now.  Thanks for the bug report, and reopen this if I've messed up!

closed 2016-09-16 by peterbarker