ArduPilot/pymavlink #4: No module named 'mavexpression'

dlion, 2016-09-17:
Hi, I followed this quickstart: http://python.dronekit.io/guide/quick_start.html
But when I type: python hello.py this is the result:
```
tarting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Traceback (most recent call last):
  File "hello.py", line 6, in <module>
    from dronekit import connect, VehicleMode
  File "/usr/lib/python3.5/site-packages/dronekit/__init__.py", line 42, in <module>
    from pymavlink import mavutil, mavwp
  File "/usr/lib/python3.5/site-packages/pymavlink/mavutil.py", line 10, in <module>
    import select, mavexpression
ImportError: No module named 'mavexpression'
```
I installed everything so I don't understand why.
I use Arch Linux, thanks

peterbarker, 2016-09-20:
This is a python3 thing.

Please try "python2 hello.py" instead.

peterbarker, 2016-09-22:
@dlion Have you resolved your issue with 'mavexpression' not found?

dlion, 2016-09-23:
@peterbarker  Sorry for the delay, with python2 hello.py it returns:
```
python2 hello.py         
Start simulator (SITL)
Traceback (most recent call last):
  File "hello.py", line 2, in <module>
    import dronekit_sitl
ImportError: No module named dronekit_sitl
```
But when I do:
```
pip install dronekit-sitl
Requirement already satisfied (use --upgrade to upgrade): dronekit-sitl in /usr/lib/python3.5/site-packages
Requirement already satisfied (use --upgrade to upgrade): psutil>=3.0 in /usr/lib/python3.5/site-packages (from dronekit-sitl)
Requirement already satisfied (use --upgrade to upgrade): dronekit>=2.0.0b6 in /usr/lib/python3.5/site-packages (from dronekit-sitl)
Requirement already satisfied (use --upgrade to upgrade): six>=1.10 in /usr/lib/python3.5/site-packages (from dronekit-sitl)
Requirement already satisfied (use --upgrade to upgrade): pymavlink==2.0.6 in /usr/lib/python3.5/site-packages (from dronekit>=2.0.0b6->dronekit-sitl)
Requirement already satisfied (use --upgrade to upgrade): monotonic==1.2 in /usr/lib/python3.5/site-packages (from dronekit>=2.0.0b6->dronekit-sitl)
```

ardupilotintegration, 2016-09-23:
TIP: if you are using a machine with both 2.x and 3.x versinos of python installed, you need to be cautious as to which version of python you execute, AND which version of pip you execute.    
in most cases, it can be clarified by using 'python2' and 'pip2' together , or 'python3' and 'pip3' together.    it looks to be like you've tried to use python2, and 'pip' ( which is defaulting to python3 )

davidbuzz, 2016-09-23:
TIP: if you are using a machine with both 2.x and 3.x versions of python installed, you need to be cautious as to which version of python you execute, AND which version of pip you execute.

in most cases, it can be clarified by using 'python2' and 'pip2' together , or 'python3' and 'pip3' together. it looks to be like you've tried to use python2, and 'pip' ( which is defaulting to python3 )

dlion, 2016-09-23:
@davidbuzz Thanks!

closed 2016-09-23 by dlion

ijustwant, 2016-10-29:
I also get this error:
ImportError: No module named 'mavexpression'

I'm running windows 7 and only Python3 installed. Changed the hello.py to version 3 print commands, but still no luck. 

Any suggestions ?

khancyr, 2016-10-31:
@ijustwant 
You need to use python2 for now