ArduPilot/pymavlink #25: Error module future

maellebranco, 2016-10-13:
Hi,
When building sitl:

Traceback (most recent call last):
  File "/home/x/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/x/ardupilot/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future

cclauss, 2016-10-13:
$ `pip install future`  # https://pypi.python.org/pypi/future or http://python-future.org/

rmackay9, 2016-10-14:
It could be related to this issue if you're building on Windows and this issue also includes some instructions for how to install "future". https://github.com/ArduPilot/ardupilot/issues/4966

maellebranco, 2016-10-14:
Thanks !
I reported it as I figured it should be automatic (which is handled with the other issue)

closed 2016-10-14 by maellebranco

tribbloid, 2016-11-07:
affect me as well, why don't you simply put future > 0.16.0 in your setup or requirement file?

khancyr, 2016-11-07:
@tribbloid it is already in , but the header generation is done by setup before installing future, and it needs future ...

tuloski, 2017-12-21:
For windows users with python 2.7.9+:
`python -m pip install -U pip`
then add C:\PythonXX to the environment variable PATH, then
`pip install future`

bys1123, 2019-04-09:
I'm using ubuntu 18.04, I got this error after run install script then tried to build.

Hasi123, 2019-04-16:
I had the same problem in ubuntu and it wouldn´t let me install future without root. The solution was:
`sudo -H pip install future`

StephenCarlson, 2021-06-18:
Its back!

I'm in Ubuntu 18.04, just git cloned the ArduPilot repo from GitHub, and I'm trying to use the waf build system. I did the configure step for the board, and then `.waf plane`, and I get the following:

``` bash
Build failed
Traceback (most recent call last):
  File "/home/steve/dev/ardupilot/modules/waf/waflib/Task.py", line 348, in process
    ret = self.run()
  File "Tools/ardupilotwaf/mavgen.py", line 55, in run
    from pymavlink.generator import mavgen
  File "/home/steve/dev/ardupilot/modules/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
```

I've done every variant of `pip install future`, include the most recent post above, `sudo -H pip install future`. Not working. Exactly the same behavior as an hour ago.

Help?

StephenCarlson, 2021-06-18:
Actually....... nevermind. The following command did it:

`python2 -m pip install future`

I had executed every suggestion in this thread and from other sources... except that variant, as I had assumed that the alias for `python` was to 2.17.x, as verified by the command `which python` and `python -V` showing the 2.17 version. So, what is going on, what is the difference between `python` and `python2`? `which python2` just points to the same python as `which python`. Anyway, it is compiling for me now.

cclauss, 2021-06-18:
Python 2 died on more than 500 days ago on 1/1/2020 and should be considered an insecure platform.

Current Python is v3.9.

julled, 2022-04-04:
If you encounter this in the context of the ros based mavlink package.
Enforce usage of python3 helped for me:

`export ROS_PYTHON_VERSION="3"`