ArduPilot/pymavlink #38: pip 2.0.9 install was failing

jgoppert, 2016-11-16:
I took down the 2.0.9 package as it won't build with pip. 2.0.8 works.

khancyr, 2016-11-16:
@jgoppert hi, Could you detail the error ?

jgoppert, 2016-11-16:
Started out that it couldn't find future. After I tried to fix that (once by using master and once by doing it myself), the generated python files weren't being included. I tried to fix it, but gave up for now.

khancyr, 2016-11-16:
Yes, new dependencies are future, lxml and python-dev .
Normally dependencies should have been solved in master but no new version has been made and it has been update on mavlink repo.

jgoppert, 2016-11-16:
@khancyr I tried updating it using master. All packaging attempts failed (2.0.10, 2.0.11, 2.0.12). I gave up at that point. The last issue I ran into was that the include path for mavnative needed to be absolute since pip was not building the extension in source. Probably best to bump the next version to 2.0.13 when you get it working. Pypi was updating slowly so I didn't realize my 2.0.9.post0 actually worked as well. So if you are stuck in a packaging nightmare like I was maybe just start doing 2.0.13.post0, 2.0.13.post1 etc. until you get it working.

cclauss, 2017-09-29:
https://pypi.python.org/pypi/pymavlink is at v2.2.5...  Can this issue be closed?

cclauss, 2018-01-10:
https://pypi.python.org/pypi/pymavlink is at v2.2.7... Can this issue be closed?

closed 2018-10-19 by amilcarlucas