ArduPilot/pymavlink #41: Pip install fails

julianoes, 2016-12-02:
I have no luck installing the package as described in the README:

```
sudo pip install .
Processing /home/julianoes/src/pymavlink
Requirement already satisfied (use --upgrade to upgrade): future in /usr/lib/python3.5/site-packages (from pymavlink==2.0.9)
Requirement already satisfied (use --upgrade to upgrade): lxml in /usr/lib/python3.5/site-packages (from pymavlink==2.0.9)
Installing collected packages: pymavlink
  Running setup.py install for pymavlink ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-jk12x8xn-build/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-ted1c5dm-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    No XML message definitions found
```

This happens on Python2 and Python3.
I also tried to install pymavlink as part of mavlink (as a submodule) and saw the same error.

OXINARF, 2016-12-03:
It is clear that it hasn't found the XML files. The code is simple:

````
if os.getenv("MDEF",None) is not None:
        mdef_paths = [os.getenv("MDEF")]
    else:
        mdef_paths = [os.path.join('..', 'message_definitions'),
                      os.path.join('mavlink', 'message_definitions'),
                      os.path.join('..', 'mavlink', 'message_definitions'),
                      os.path.join('message_definitions'),
        ]
```

So you either have a wrong MDEF env var set or you don't have the files in any of those directories.

julianoes, 2016-12-04:
@OXINARF thanks for your help, appreciated.

I didn't know about the `MDEF` env variable but I can see that it should not be necessary to set it for my case because the definitions can be found in `../message_definitions`.

I realized that `sudo python setup.py install` works, however `sudo pip install .` does not.

khancyr, 2016-12-04:
Yep that a problematic, it is because pip install is excecuted from /tmp not from pymavlink directory. I will works on a patch and I need to finish the readme correctly

julianoes, 2016-12-04:
@khancyr yes you're right, it's because it moves to /tmp. Also, I tried to use MDEF but it gets lost by `sudo`. This finally worked:

```
sudo MDEF=$HOME/src/mavlink/message_definitions pip install .
```

julianoes, 2017-03-29:
Thanks @khancyr, closing.

closed 2017-03-29 by julianoes