ArduPilot/pymavlink #43: Always False condition in mavlink_parse_char() in Parser.java

ixig, 2017-01-04:
```
  public MAVLinkPacket mavlink_parse_char(int c) {
        msg_received = false;
        switch (state) {
            ...
            case MAVLINK_PARSE_STATE_GOT_STX:
                if (msg_received) {
                    msg_received = false;
                    state = MAV_states.MAVLINK_PARSE_STATE_IDLE;
                }
```

Since msg_received is initialized to false just before the switch statement, the test of msg_received in the case statement will always be false, given the preceding cases always break out.

WickedShell, 2017-01-04:
Absolutely correct, we should fix that, we would never want to go into that case either as far as I can tell. The good news is the JIT probably fixed it for us, but its still worth removing it from the code anyways. (as well as the unneeded sets to false)

WickedShell, 2017-01-12:
@ixig #44 addresses this, I don't actually see any reason to use the msg_received switch at all :)

ixig, 2017-01-12:
Excellent ... On a different note, will you entertain a pull request to fix (Android) Lint recommendations for stuff like replacing for's with foreach, and adding 'finals', trying to print arrays in the toString(), etc. as appropriate? Due to the sheer number of classes auto-generated, these can add up quickly and have a big impact on users who are trying to keep a 'clean' project.

WickedShell, 2017-01-12:
Do you have some examples of what things you think are should be adjusted?
I don't work with Android at all (or for that matter run a Java linter over
the bindings...) so I've never seen a warning on any of that.

WickedShell, 2017-01-12:
Closing as the fix for this was merged, but feel free to open issues for the for/foreach stuff. I did run findbugs last night and fixed the things it labeled as performance impairments, but I haven't gotten them tested yet.

I'm definitely in favor of clean projects, and as long as it doesn't destroy readibility/performance seems quite reasonable.

closed 2017-01-12 by WickedShell

ixig, 2017-01-25:
Here's a list of common lint warnings reported by Android Lint (mostly on the msg_*.java):

- _Declaration can have final modifier_ => any ML field that's an array (e.g int x[] = new int[...];)
- _'for' loop replaceable with 'foreach'_ => in msg.pack() anytime there's a ML field that's an array
- _Declaration has Javadoc problems_ => '@return' tag description is missing
- _Call to 'toString()' on array_ (this one is classified as Probable Bug) => in msg.toString() when there's a ML field that's an array

WickedShell, 2017-01-25:
Yeah I saw all the toString on array ones, I need to dive deeper into the generator to sort out the array stuff.

Regarding the final modifier on the fields, I think that works for both of my use cases, but I think that could break other users, which is why I was avoiding it. For example the PARAM_VALUE packet has a public byte param_id[] = new byte[16]; array, as far as I can tell an absolutely valid usage is to actually externally just do foo.param_id = "MY_NEW_PARAM"; Rather embarrassingly I only just noticed that there were get/set helpers for all the arrays, for some reason all my code uses (System/arraycopy ...). Am I missing something here where this would actually be a harmless change?

I did push some additional fixes that findbugs pointed out to me. https://github.com/ArduPilot/pymavlink/commit/d1fcbc47fbed1d93e379c0c6e3d0e0bfc73fd4e6

wpoon-gpfw, 2017-01-26:
Opps, yes you're correct on the _final_ being inappropriate for the array public fields! In my application, I wasn't creating packets, so lint made that overzealous recommendation.

WickedShell, 2017-01-26:
@wpoon-gpfw I mean it's not entirely unreasonable to say that you need to set all the arrays with the set%s() methods, and make the backing array private and final, but it would be a function change from how it is now, and would potentially break existing users code, and every other field is public without a getter/setter requirement, so I'm inclined to say we shouldn't do that change now.