ArduPilot/pymavlink #56: Current version available on PyPl is confused

courtarro, 2017-03-21:
When I query for the current version of `pymavlink` on PyPl (using the site or `pip`), I get `2.0.6`. However, when I install the package, I end up with `2.1.0`.

Apparently `2.1.0` was uploaded on 2017-02-02, then `2.0.6` was uploaded on 2017-02-06. As a result, either PyPl is confused or `pip` is. This page shows that `2.0.6` is the latest version:

https://pypi.python.org/pypi/pymavlink

However, check this page to see the confusion:

https://pypi.python.org/pypi/pymavlink/2.1.0

That's the page for version `2.1.0`, but it shows a "Latest version" of `2.0.6`. Was `2.0.6` reuploaded on purpose, perhaps to deal with a regression? If so, `2.1.0` is what's actually being downloaded by my `pip` instance. I'm not sure why, other than it looking at the version numbers and trying to get the "highest" one. This is what I got:

    root@docker:/# pip install pymavlink
    Collecting pymavlink
      Downloading pymavlink-2.1.0.tar.gz (2.8MB)
        100% |################################| 2.8MB 552kB/s

paan2097, 2017-04-27:
Could it be that pymavlink 2.1.0 has problems?? Because I'm trying to run dronekit scrips with pymavlink 2.1.0 and I'm getting link timeouts.....could this be the problem?

cclauss, 2017-04-28:
https://pypi.python.org/pypi/pymavlink 2.2.2 is current.  Can you please try that version and then describe the problems that remain?

Thanks, CCC

closed 2017-04-28 by courtarro