ArduPilot/pymavlink #68: Installing pymavlink on Ubuntu 16.04 fails

ufechner7, 2017-04-08:
I tried to install pymavlink with pip:
sudo pip2 install -U pymavlink
No error message.
Trying to import mavutil 
import mavutil
first gives the error message "module not found"
If I add the following line to my .bashrc file:
export PYTHONPATH="${PYTHONPATH}:/usr/local/lib/python2.7/dist-packages/pymavlink"
and try to import mavutil I get a different error message:
`In [1]: import mavutil
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-1-062bfe057715> in <module>()
----> 1 import mavutil

/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py in <module>()
    105 
    106 # Set the default dialect. This is done here as it needs to be after the function declaration
--> 107 set_dialect(os.environ['MAVLINK_DIALECT'])
    108 
    109 class mavfile(object):

/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py in set_dialect(dialect)
     80     '''
     81     global mavlink, current_dialect
---> 82     from .generator import mavparse
     83     if 'MAVLINK20' in os.environ:
     84         wire_protocol = mavparse.PROTOCOL_2_0

ValueError: Attempted relative import in non-package
`
Any idea?

ufechner7, 2017-04-09:
Ok, I got a step further:
```
I installed pymavlink with the following commands:

sudo apt-get install libxml2-dev libxslt-dev python-dev
sudo pip2 install -U future lxml
sudo pip2 install -U pymavlink

I added the following line to .bashrc:
export PYTHONPATH="${PYTHONPATH}:/usr/local/lib/python2.7/dist-packages/pymavlink"

This works:
from pymavlink import mavutil

cd /usr/local/lib/python2.7/dist-packages/pymavlink
sudo mavgen.py -o mavlinkv10.py message_definitions/v1.0/common.xml

joe mavlinkv10.py
--> in line 13, replace ... with .
--> save

from pymavlink import mavlinkv10 as mavlink

is working now.
```
Is this complicated way that includes manual editing of mavlinv10.py the intended way to use this software?

crashmatt, 2017-08-25:
I have success with 16.04 by installing from source

python setup.py install --user

If you have installed libraries already as a sudo install then they will take priority over the user installation.  You will need to do a python setup.py remove or similar to remove those libraries first.

closed 2018-10-19 by amilcarlucas