ArduPilot/pymavlink #69: Unable to install pymavlink on Ubuntu 16.04

TiagoSanLou, 2017-04-18:
I successfully installed all the dependencies, but when trying to install with `sudo pip2 install -U pymavlink` I get:

> Collecting pymavlink
  Using cached pymavlink-2.2.0.tar.gz
Requirement already up-to-date: future in ./.local/lib/python2.7/site-packages (from pymavlink)
Requirement already up-to-date: lxml in ./.local/lib/python2.7/site-packages (from pymavlink)
Building wheels for collected packages: pymavlink
  Running setup.py bdist_wheel for pymavlink ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-D1ZhFK/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpXSxTnnpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  No XML message definitions found
  
  ----------------------------------------
  Failed building wheel for pymavlink
  Running setup.py clean for pymavlink
Failed to build pymavlink
Installing collected packages: pymavlink
  Running setup.py install for pymavlink ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-D1ZhFK/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-ITjw7D-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    No XML message definitions found
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-D1ZhFK/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-ITjw7D-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-D1ZhFK/pymavlink/

Any help would be appreciated!

WickedShell, 2017-04-18:
@peterbarker this appears to be associated with the Travis error we were looking at on mumble.

peterbarker, 2017-04-18:
tridge has uploaded a new package which fixes this.

closed 2017-04-18 by peterbarker

TiagoSanLou, 2017-04-19:
Timely responded! Thanks!