ArduPilot/pymavlink #71: Would it be possible to start tagging releases?

cclauss, 2017-04-22:
Would it be possible to start using https://github.com/ArduPilot/pymavlink/releases ?  Or at least a changelog?

Also this repo needs a __license file__.

ferret-guy, 2017-06-29:
I wanted to follow up on this, I think it would be useful. In addition, this would enable automatic PyPI releases with tagged commits.

@tridge

ferret-guy, 2017-06-29:
On the Licence front, I agree we need one; it should probably be LGPL. There is plenty of conflicting information, in [setup.py](https://github.com/ArduPilot/pymavlink/blob/master/setup.py) it has `license='LGPLv3'` but there are plenty of files like [pymavlink/mavexpression.py](https://github.com/ArduPilot/pymavlink/blob/master/mavexpression.py) which has the text `Released under GNU GPL version 3 or later` that is incompatible, but it looks like they were all committed by @tridge so conceivably he could move it to GPL.

cclauss, 2018-01-10:
Original (Would it be possible to start using https://github.com/ArduPilot/pymavlink/releases ?  Or at least a changelog?) was a duplicate of #6

amilcarlucas, 2018-10-19:
@tridge this one should be easy to fix and it is important

khancyr, 2018-10-19:
there is already a license file : COPYING 
but we should address the licensing on file issue

closed 2019-03-07 by cclauss