ArduPilot/pymavlink #74: Problems parsing logs from APM 3.4.6: no gps data found

larsks, 2017-05-04:
I'm trying to use some of the pymavlink utilities to parse [this log file](http://chunk.io/f/7a175259fa244ca58f5bd63979dde240).  Running `mavsummarize.py` results in:

    Warning: No GPS data found, can't give position summary.

But running `mavlogdump.py` on the same file yields what appear to be GPS records, e.g:

    2017-05-03 12:28:46.00: GPS {TimeUS : 80221069, Status : 3, GMS : 318541000, GWk : 1947, NSats : 15, HDop : 0.76, Lat : 42.3844842, Lng : -71.1628935, Alt : 16.77, Spd : 0.0608276240528, GCrs : 170.53767395, VZ : 0.00999999977648, U : 1}

And other utilities -- e.g., `mavtogpx.py` -- seem to have no problem finding position data in the log.  Is there something wrong with the log? With the utilities?

peterbarker, 2017-05-04:
On Wed, 3 May 2017, Lars Kellogg-Stedman wrote:

> And other utilities -- e.g., mavtogpx.py -- seem to have no problem finding
> position data in the log. Is there something wrong with the log? With the
> utilities?

mavsummarize.py only understands messages from tlog files.

Adjusting it for BIN files will involved understanding (at minimum):
  - GPS messages
  - MODE messages
  - MSG messages (to get the vehicle type)
  - how to map the mode number in MODE to whether the mode is autonomous or 
not

amilcarlucas, 2020-09-24:
@peterbarker how about closing this as WONT_FIX ?

peterbarker, 2020-09-24:
@amilcarlucas I'm convinced.

closed 2020-09-24 by peterbarker