ArduPilot/pymavlink #77: Mavutil / PX4: 'interpret_px4_mode' returns a mode not mapped in 'px4_map'

epinal-animusoft, 2017-05-08:
Hi guys,

I'm having this issue in mavutil.py:

'interpret_px4_mode' function is returning a not mapped mode in 'px4_map' (LAND) which actually exists based on this: https://docs.px4.io/en/flying/flight_mode_selection.html

Example:
```
# Heartbeat Message:
m = {'type' : 13, 'autopilot' : 12, 'base_mode' : 29, 'custom_mode' : 100925440, 'system_status' : 3, mavlink_version : 3}

mode = mavutil.interpret_px4_mode(base_mode, custom_mode)  # mode = 'LAND'
mode_is_valid = mode in mavutil.mode_mapping()  # False 'LAND' not in px4_map
```
I think the solution could be adding all the missing modes to px4_map.
Let me know how to proceed.

peterbarker, 2017-05-09:
@squilter @sanderux thoughts?

sanderux, 2017-05-10:
LAND mode needs to be added here:
https://github.com/ArduPilot/pymavlink/blob/6657fe16f0135ab55e8b34247c90e00688ea0050/mavutil.py#L1552-L1562

peterbarker, 2017-05-10:
@epinal - @sanderux has been good enough to supply a patch for this issue.  Are you in a position to test his fix?

You can find the patch here: https://github.com/ArduPilot/pymavlink/pull/80

epinal, 2017-05-11:
Sure I'll test it and let you know ASAP. Thanks guys

epinal, 2017-05-11:
Tested. Now it's working.
Thank you guys @peterbarker @sanderux

closed 2017-05-11 by epinal