ArduPilot/pymavlink #78: Decode method behaviour.

edwinhayes, 2017-05-09:
What is the purpose of the method `MAVLink.decode(...)`?  This code fails:

```
msg = mav.heartbeat_encode(0,0,0,0,0,0)
buf = msg.pack(may, force_mavlink1 = True)
msg2 = mav.decode(buf)
```

But succeeds if the last line is replaced with:

```
msg2 = mav.parse_char(c)
```

It appears the decode method fails to consider which MAVLink version the message contained in the buffer is?

Firstly, it seems totally unintuitive you can pass multiple characters at once to `parse_char` - the only clue in the documentation is that the method supposedly takes 'bytes' plural.  Secondly, what then is the purpose of the decode method, if it isn't to decode a buffer of MAVLink bytes?

cclauss, 2017-05-09:
This is a difference between Python 2 and Python 3.  Bytes is a builtin class in Python 3 which has a [bytes.decode()](https://docs.python.org/3/library/stdtypes.html?highlight=decode#bytes.decode) method.  Python 2 has no such constructs and always dangerously mix bytes, ascii strings, and Unicode strings.  Bytes/strings issues are usually the most difficult part of creating a codebase that supports both versions of Python.

edwinhayes, 2017-05-09:
Ah, right.  We're on Python 2.7

So, based on a cursory glance, looks like parse_char() converts the buffer to fix the encoding, then called decode()?  This then basically seems like an API issue: why is decode() public/documented if the appropriate method for decoding messages is parse_char()?