ArduPilot/pymavlink #90: future module error

medhijk, 2017-07-05:
While running the sim_vehicle.py I'm getting this error message.

[4/4] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Traceback (most recent call last):
  File "/home/medhijk/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/medhijk/ardupilot/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
**ImportError: No module named future**

mavgen returned 1 error code
Waf: Leaving directory `/home/medhijk/ardupilot/build/sitl'
Build failed
 -> task in 'mavlink' failed (exit status 1): 
	{task 139927157712448: mavgen ardupilotmega.xml -> }
''
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks

I thought it's missing the future module and so tried installing future by this command: 
**pip install future**
But I get a message from the terminal as:
**Requirement already satisfied: future in /usr/local/lib/python3.5/dist-packages**

Can anyone suggest a way out of this?

khancyr, 2017-07-05:
Hello, 
Use pip2 install future

medhijk, 2017-07-05:
Thanks, that works!

closed 2017-07-05 by medhijk

WaldenLeefx, 2022-01-21:
@khancyr Thank you very much, it works!!!