ArduPilot/pymavlink #97: fatal: memory overflow if length=0 with limited buffer size

az0uz, 2017-07-28:
In the C API v2.0, helper receive function `mavlink_frame_char_buffer`:
file: `generator/C/include_v2.0/mavlink_helpers.h`

In case of a glitchy packet with a `len` field value of `0` (or if you would have defined a message of length zero), and MAVLINK_MAX_PAYLOAD_LEN smaller than 249 (it is rounded to the next 8 bytes alignement, not sure if that's a good idea, so effectively 256 bytes):

The state machine doesn't handle correctly a null length:
In my case this happens because of a glitchy reception that changed the `len` field to zero (I don't have any zero-length messages) which happens every few minutes given my external conditions.

when it get to the state `MAVLINK_PARSE_STATE_GOT_MSGID3` (line 714)
The code read the first byte even if `len==0` and test `status->packet_idx == rxmsg->len` only after reading this first byte witch will be only true after reading 256 bytes instead of 0.

if you used `MAVLINK_MAX_PAYLOAD_LEN` to reduced the size of the buffer, their will be bytes overriding an unknown segment of memory witch leads to undefined behavior.

In my case using it on a atmega mcu it totally freezes the execution.

peterbarker, 2017-07-30:
On Fri, 28 Jul 2017, Antoine de Maleprade wrote:

> In the C API v2.0, helper receive function mavlink_frame_char_buffer:
> file: generator/C/include_v2.0/mavlink_helpers.h
>
> In case of a glitchy packet with a len field value of 0 (or if you would
> have [...]

Nice find.  Do you have a handy tlog which replicates?

az0uz, 2017-07-31:
No I haven't any, I am running the code on an atmega32u4 MCU without real debugging tools but I managed to track down what was wrong using Serial and I fixed the issue, You can review Merge Request #98.
To reproduce, set MAVLINK_MAX_PAYLOAD_LEN to, say, 64 bytes, and send any message but replace the length field by 0.

commit daad13d6 (2018-01-10): C_API_v2.0: skip payload read when len is null ( fix: #97 )
closed 2018-01-10 by tridge