ArduPilot/pymavlink #102: Bug in mavgen_wlua.py while trying to create a Mavlink Wireshark dissector(?)

saihv, 2017-08-15:
I was trying to create a mavlink Wireshark dissector through mavgen (and consequently mavgen_wlua.py) and noticed that there is a problem in the lines 250-257 section. 

` if (unknownFrameBeginOffset ~= 0) then
            pinfo.cols.info:append("Unkown message22")
            size = offset - unknownFrameBeginOffset
            subtree:add(f.rawpayload, buffer(unknownFrameBeginOffset,size))
            unknownFrameBeginOffset = 0
            -- jump to next loop
            break
        end
`

Essentially, LUA complains that unknownFrameBeginOffset was a variable that was not 'declared' in a global scope prior to this use, and throws an error saying `attempt to perform arithmetic on a global nil value: unknownFrameBeginOffset`. Once I set this variable to zero before the the message parsing and version checks (before line [205](https://github.com/ArduPilot/pymavlink/blob/master/generator/mavgen_wlua.py#L205)), it works as expected.

tatsuy, 2018-01-29:
Great! I could parse MAVLink messages. Would you submit this change?

pietrodn, 2018-03-07:
This issue is also mentioned in this forum: https://diydrones.com/forum/topics/mavlink-lua-dissector

commit 19473add (2018-03-13): Fix unknownFrameBeginOffset bug in mavgen_wlua.py

Without unknownFrameBeginOffset = 0 being set, Wireshark isn't able to
decode any MAVLink message

Fixes #102
closed 2018-03-13 by peterbarker