ArduPilot/pymavlink #107: mavutil : Ignores target component/system when processing messages

crashmatt, 2017-08-25:
mavutil won't work with multiple heartbeats/systems in a stream since it doesn't filter for the target system.  It stores received messages regardless of the source.  It processes system status, gps, and parameters regardless of source.

Possible code locations where a filter might be implemented.

in recv_mesg here:
https://github.com/ArduPilot/pymavlink/blob/master/mavutil.py#L320
and again here:
https://github.com/ArduPilot/pymavlink/blob/master/mavutil.py#L918

In post_message ;
https://github.com/ArduPilot/pymavlink/blob/master/mavutil.py#L224

The signature checking turns up a little late to the party and starts checking after the fact:
https://github.com/ArduPilot/pymavlink/blob/master/mavutil.py#L288

A simple fix is to check message source against self.target_system and self.target_component.  If targets are zero then let the message through anyway.  This will allow heartbeat to be set and seen.

Additional problem with wait_heartbeat and anything else using recv_match:
recv_match uses recv_msg.  recv_msg does different things depending on the master type.  One is to post_message, another is to store message directly etc..
The behaviour of recv_match changes depending on situation.  For example: at startup you may wish to wait for any heartbeat or a specific heartbeat.

peterbarker, 2017-08-26:
On Fri, 25 Aug 2017, Matthew Coleman wrote:

Have you looked at my mavsource branch?

https://github.com/peterbarker/pymavlink/tree/mavsource

crashmatt, 2017-08-26:
I had not but I had a quick look now.  I see the interesting "listener"
idea in your last commit.  What does your branch do and what is your plan?

A few thoughts:
I have multiple heartbeats in one stream.  The majority of the time I wish
to interface with only one of the heartbeats/systems.
My main plan is to have a core instance of mavproxy which only forwards
packets to mavproxy instances which each consume messages from specific
targets.  This way I only need to download the parameters for the systems I
am interested in.

Different filters are required depending on what is using the messages.
The proxy might choose to filter messages before forwarding.   This would
lower total processing overhead.  This would require more configuration
fields in mavproxy which may not be so easy to use.
Global filtering at input prevents any modules from receiving data from the
wrong target.  This causes problems if a module needs data from all targets
ie. mapping.  Maybe listener solves this.

Right now the mavlink hops between the target system it uses depending on
the last heartbeat it got.  Then it downloads params from a random target.
Just fixing this is a good start.
mavlink has much specific functionality bundled together with core
library.  I understand why this has happened and that it is gradually
improving.  If the specifics were all cleaned into modules then this would
all be an easier task.

And finally: If changes are too big or have too much risk then it never
makes it into upstream.  Whatever is done needs to be stealthy ie. it
doesn't change the default functionality of either mavproxy or mavlink.

peterbarker, 2017-08-26:
On Sat, 26 Aug 2017, Matthew Coleman wrote:

> I had not but I had a quick look now. I see the interesting "listener"
> idea in your last commit. What does your branch do and what is your plan?

It simply allows you to select a single sourceid/componentid to listen to. 
This avoids several problems within pymavlink, most notably waypoint 
handling.  It is designed to not bother existing users.

You can see it in use in my dronekit-python avoidance work.

> And finally: If changes are too big or have too much risk then it never
> makes it into upstream. Whatever is done needs to be stealthy ie. it
> doesn't change the default functionality of either mavproxy or mavlink.

Well, yes.  I haven't convinced tridge this is a good one yet.  He 
indicated he'd like to see mAVProxy reimplemented in terms of mavsource to 
show it was general enough.  Turns out it isn't a mavsource would need to 
take from several streams to provide the redundant-link functionality....

crashmatt, 2017-08-26:
Does it forward messages from all sources?  If so, this is exactly what I
need.

Can you give me an idea of how to use it?  Does it use the mavproxy
target_system/component?

There is lots of thing that could be more ideal than they are. A
re-implementation of mavroxy will not happen quickly. From my point of view
a smaller change would improve something fundamentally broken beyond any
requirements for link redundancy.  Maybe I am missing the point.

peterbarker, 2017-08-26:
On Sat, 26 Aug 2017, Matthew Coleman wrote:

> Can you give me an idea of how to use it? Does it use the mavproxy
> target_system/component?

https://github.com/peterbarker/dronekit-python/blob/source-system-filtering/examples/avoidance/avoidance-manytests.py#L409

Here's a class which inherits from mavsource:

https://github.com/peterbarker/dronekit-python/blob/source-system-filtering/dronekit/mavlink.py#L41

crashmatt, 2017-08-27:
Submitted a PR which has the simplest solution I could find.  Quick and dirty but effective enough.  Very low risk on existing functionality.

[pymavlink PR submitted here](https://github.com/ArduPilot/pymavlink/pull/111)
[MAVProxy PR submitted here](https://github.com/ArduPilot/MAVProxy/pull/444)

amilcarlucas, 2018-10-19:
Now that PRs are available I'm closing this

closed 2018-10-19 by amilcarlucas