ArduPilot/pymavlink #123: MAVLink v2.0 support for Java generated library

Hunter522, 2017-11-06:
The Java generated library has no support for MAVLink v2.0.

seeul8er, 2017-11-15:
MAVLink v2 would be really nice indeed :+1:

Hunter522, 2017-11-15:
I began working on this. Making progress. Will make a pull request hopefully in the next couple weeks.

WickedShell, 2017-11-16:
If you are looking at the Java parser at a fundamental level it might be worth keeping in mind that uint64_t is decoded incorrectly at the moment. Unfortunately the only solution I've found for this is to decode it into a BigInteger, which results in a lot of unintentional breakage/API changes for all current users which is why I didn't do this. I'm not sure if it's worth the effort of doing it now, or just continue to rely on all users who want to use a uint64_t needing to be aware of large values becoming negative. Just mentioning it as food for thought.

gmorph, 2017-11-16:
Thanks for working on this.  I too am keen to see this done as we can then incorporate it into Tower and have it support MAVLink 2.0 messages.  Thanks.

amilcarlucas, 2021-02-22:
I think this is done now and can be closed.