ArduPilot/pymavlink #126: DFReader recv_msg() returns NANs

fnoop, 2017-11-29:
eg:
```POS {TimeUS : 19429266, Lat : -22.2772117, Lng : -51.5002036, Alt : 0.0299999993294, RelHomeAlt : 0.0310829151422, RelOriginAlt : nan}```

Fields with nan should possibly not be set, or set to None/False, or perhaps it indicates an internal dfreader/pymavlink bug that could be fixed?

peterbarker, 2017-11-29:
On Wed, 29 Nov 2017, Fnoop wrote:

> eg:
> POS {TimeUS : 19429266, Lat : -22.2772117, Lng : -51.5002036, Alt :
> 0.0299999993294, RelHomeAlt : 0.0310829151422, RelOriginAlt : nan}
> 
> Fields with nan should possibly not be set, or set to None/False, or perhaps
> it indicates an internal dfreader/pymavlink bug that could be fixed?

Nope.  Working as intended :-)

There was a bug where we were getting NANs in that field.

Now we do it on purpose.  You will get a NAN in that field when no origin 
has been set.  Still useful to get the other information in the POS 
message, so we still emit it.

It *should* be a "quiet nan" now-adays - depends on your operating system 
generating the log as to whether that works.  That will appear as "qnan".

In terms of not-set and true/false - there's no out-of-band actual-number 
in this case.

Incidentally, if you want to test your own dataflash reading code, see the 
sample output:

pbarker@bluebottle:~/rc/ardupilot/libraries/DataFlash/examples/DataFlash_AllTypes(offboard-auth)$ 
git log output.BIN
commit 6536bf39b33e4d4def8d247eb0639ef0792e489e
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Sat Jul 22 12:29:05 2017 +1000

     DataFlash: add DataFlash_AllTypes sample output
pbarker@bluebottle:~/rc/ardupilot/libraries/DataFlash/examples/DataFlash_AllTypes(offboard-auth)$

amilcarlucas, 2018-04-13:
I think this one can be closed now.

closed 2018-04-13 by amilcarlucas