ArduPilot/pymavlink #130: Memory leak in generated dialect code

SamuelDudley, 2017-12-06:
Hi,
While working on the stand alone MAVCesium server @fnoop had identified that memory resources were continuing to be used. I eventually tracked down the cause to this code segment within pyMAVLink:
https://github.com/ArduPilot/pymavlink/blob/master/generator/mavgen_python.py#L526

[image]

Issue where the problem was identified: https://github.com/SamuelDudley/MAVCesium/issues/34

The behavior we observed was the result of a bytearray being extended over and over again, taking up more and more memory over time.

The issue is that it is quite possible to __always__ have a complete message waiting to be decoded when `parse_char()` is called. In this case the final `else` statement of  `parse_char()` never gets ran and the buffer continues to grow.

```
        def parse_char(self, c):
            '''input some data bytes, possibly returning a new message'''
            self.buf.extend(c)
            self.total_bytes_received += len(c)
            .....
            .....
            if m != None:               # <---- this always returns true with a new message.
                self.total_packets_received += 1
                self.__callbacks(m)
            else:                            # <---- so this code never gets a chance to run, causing a memory leak.
                if self.buf_len() == 0 and self.buf_index != 0:
                    self.buf = bytearray()
                    self.buf_index = 0
            return m
```
Although the logic in that statement `self.buf_len() == 0 and self.buf_index != 0` is true every time this method is called, `self.buf` and `self.buf_index` are never able to be reset... personally I think the else could be removed and we should be testing that condition every time that method is called. e.g.:
```
        def parse_char(self, c):
            '''input some data bytes, possibly returning a new message'''
            self.buf.extend(c)
            self.total_bytes_received += len(c)
            .....
            .....
            if m != None:
                self.total_packets_received += 1
                self.__callbacks(m)
            if self.buf_len() == 0 and self.buf_index != 0:
                self.buf = bytearray()
                self.buf_index = 0
            return m
```
Which equivalent to what is now in the MAVCesium server code: https://github.com/SamuelDudley/MAVCesium/blob/master/app/cesium_web_server.py#L215

amilcarlucas, 2018-10-19:
> 
> 
> I'm going to call it a bug... The behavior we were seeing is the result of a bytearray being extended over and over again.
> In the dialect generated code e.g. ardupilotmega.py there is a `MAVLink` object that stores the connection buffer / bytearray in `self.buf` and an associated index in `self.buf_index`.
> In the `parse_char()` method of this object the problem occurs (note I have removed some unused mavnative logic below for clarity):
> 
> ```
>         def parse_char(self, c):
>             '''input some data bytes, possibly returning a new message'''
>             self.buf.extend(c)
>             self.total_bytes_received += len(c)
>             m = self.__parse_char_legacy()
>             if m != None:
>                 self.total_packets_received += 1
>                 self.__callbacks(m)
>             else:
>                 if self.buf_len() == 0 and self.buf_index != 0:
>                     self.buf = bytearray()
>                     self.buf_index = 0
>             return m
> ```
> 
> The issue is that every time this code is run there is **always** a complete message waiting to be decoded. Because of this the final `else` statement never gets ran and the buffer continues to grow.
> Although the logic in that statement `self.buf_len() == 0 and self.buf_index != 0` is true every time this method is ran, `self.buf` and `self.buf_index` are never able to be reset... personally I think the else could be removed and we should be testing that condition every time that method is called. e.g.:
> 
> ```
>         def parse_char(self, c):
>             '''input some data bytes, possibly returning a new message'''
>             self.buf.extend(c)
>             self.total_bytes_received += len(c)
>             m = self.__parse_char_legacy()
>             if m != None:
>                 self.total_packets_received += 1
>                 self.__callbacks(m)
>             if self.buf_len() == 0 and self.buf_index != 0:
>                 self.buf = bytearray()
>                 self.buf_index = 0
>             return m
> ```
> 
> which equivalent to what is now in the MAVCesium server code.

@SamuelDudley Can you create a PR for it ?

SamuelDudley, 2018-10-19:
#131

closed 2026-02-23 by peterbarker