ArduPilot/pymavlink #133: Can you please put a ready to use Wireshark dissector lua script somewhere?

rodizio1, 2018-01-08:
I've gone through python dependency hell and back now, tried all kinds of stuff, compiling from source, using pip packages, getting all kinds of strange error messages like `File "/usr/src/mavlink/pymavlink/generator/mavgen.py", line 211, in <module>     raise DeprecationWarning("Executable was moved to pymavlink.tools.mavgen")` or   `File ./mavgen.py", line 18, in <module>   from . import mavparse ValueError: Attempted relative import in non-package`

Now after hours of tinkering I don't even get any output from mavgen.py or other commands like `python -m pymavlink.generator.mavgen --lang=wlua mymavlink.xml -o mymavlink.lua` anymore.

Apart from that, I'm totally confused by the documentation. Can you guys please, please put a ready-to-use wireshark dissector script somewhere? All I want to do is look at some wireshark traces with mavlink packets in them, not become a Python expert.

peterbarker, 2018-01-10:
On Mon, 8 Jan 2018, Rodizio wrote:

> I've gone through python dependency hell and back now, tried all kinds of
> stuff, compiling from source, using pip packages, getting all kinds of
> strange error messages like File [...]

Coincidentally I may have merged a fix for that this-arvo.

> Now after hours of tinkering I don't even get any output from mavgen.py or
> other commands like python -m pymavlink.generator.mavgen --lang=wlua
> mymavlink.xml -o mymavlink.lua anymore.

Does a clean tree work for you?

> Apart from that, I'm totally confused by the documentation. Can you guys
> please, please put a ready-to-use wireshark dissector script somewhere? All
> I want to do is look at some wireshark traces with mavlink packets in them,
> not become a Python expert.

Sorry, I've never used the lua stuff.  I believe there's a good mavlink 
inspector in one of the GCS - qgroundcontrol?

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

pietrodn, 2018-03-31:
The instructions in the file were outdated: [they are now correct](https://github.com/ArduPilot/pymavlink/commit/19473add78c0d6fd5b323aded5eaabf2d60446dd#diff-109a94c01d0771a3429a02a9a4b543e8). Can you retry?

reluctantgithubuser, 2018-11-26:
getting this error when running according to the directions:
/usr/bin/python: No module named tools

Why not just have the lua script available in the git repository to download? Does it really need to be generated each time?

amilcarlucas, 2021-02-22:
can you please test https://github.com/ArduPilot/pymavlink/pull/498