ArduPilot/pymavlink #135: Want to send some custom message, but can not find enough example.

ideaRunner, 2018-01-10:
Hi,
I want to send a custom message to my device via USBtoTTL.
I have been searching some examples for long time, but still can not finish it.
Can you give me some help or one example?
Here is my code 
`
from pymavlink.dialects.v10 import common as mavlink
from pymavlink import mavutil 

master = mavutil.mavlink_connection("/dev/ttyUSB0", 57600)
mav = master.mav
mav.command_long_send(target_system=71, target_component=67,
                      command=mavlink.MAV_CMD_DO_MOUNT_CONTROL,
                      confirmation=0,
                      param1=200,
                      param2=200,
                      param3=200,
                      param4=0,
                      param5=0,
                      param6=0,
                      param7=0)
`
and I got an output

> Device /dev/ttyUSB0 is dead

I also tried other types of codes, but not work.
Any helpful reply will be appreciated, thank you very much.

ideaRunner, 2018-01-10:
By the way, my device can just receive mavlink command.
So I did't use 
```python
master.wait_heartbeat()
```
to get the target id

khancyr, 2018-01-10:
Things to look at : 
Does /dev/ttyUSB0 exist ? (ls /dev/tty* for example)
Do you have permission to write on /dev/ttyUSB0 ? (check you are member of dialout group)

pietrodn, 2018-02-18:
Which Python version are using? I got that error message on Python 3.