ArduPilot/pymavlink #151: C# generator problem

hamishwillee, 2018-02-20:
Generating C# source via mavlink generators gives the following errors:
```
This compiler is provided as part of the Microsoft (R) .NET Framework, but only supports language versions up to C# 5, which is no longer the latest version. For compilers that support newer versions of the C# programming language, see http://go.microsoft.com/fwlink/?LinkID=533240

error CS2001: Source file 'CS\common\ByteArrayUtil.cs' could not be found
error CS2001: Source file 'CS\common\FrameworkBitConverter.cs' could not be found
error CS2001: Source file 'CS\common\Mavlink.cs' could not be found
Error
Error: Compilation failed. (1)
```
This was reported in mavlink/mavlink:
* https://github.com/mavlink/mavlink/issues/813
* https://github.com/mavlink/mavlink/issues/789
* https://github.com/mavlink/mavlink/issues/482

magicrub, 2018-02-20:
Only C was 100% ported to V2. Java is a work in progress, almost done but I think still has a few issues being worked. To my knowledge no other languages have been attempted to support V2 features.

magicrub, 2018-02-20:
I guess what I'm trying to say is we should add an immediate compile error with a useful message at the start when compiling V2 on unsupported languages.

hamishwillee, 2018-02-21:
@magicrub That is a good idea.  Are there any issue numbers that can be track the implementation of V2 on other languages?

iliasam, 2018-02-22:
I've got the same problem when I tried to generate v1 C# code.
I can resolve it by replacing line 331 of mavgen_cs.py:
`includedCsFiles =  [ 'CS/common/ByteArrayUtil.cs', 'CS/common/FrameworkBitConverter.cs', 'CS/common/Mavlink.cs'  ]`
by this code:
`    cur_dir = os.path.dirname(os.path.abspath(__file__)) + '/'
    includedCsFiles =  [cur_dir + 'CS/common/ByteArrayUtil.cs', cur_dir + 'CS/common/FrameworkBitConverter.cs', cur_dir + 'CS/common/Mavlink.cs'  ]`

hamishwillee, 2018-02-22:
@magicrub Hi Tom

> I guess what I'm trying to say is we should add an immediate compile error with a useful message at the start when compiling V2 on unsupported languages.

I updated the text. 

Who assigns work in this project? ie it would be good to get someone assigned to creating that compiler error and investigating the (unrelated) fix proposed by @iliasam above: https://github.com/ArduPilot/pymavlink/issues/151#issuecomment-367690904

amilcarlucas, 2018-10-19:
@iliasam can you please create a github pull request with that code change?

Alex-Horlock, 2019-04-23:
> I've got the same problem when I tried to generate v1 C# code.
> I can resolve it by replacing line 331 of mavgen_cs.py:
> `includedCsFiles = [ 'CS/common/ByteArrayUtil.cs', [...]

illiasam has the solution... works perfectly after implementing his fix but the initial download is still broken today (23/04/2019). Cheers.

hamishwillee, 2019-05-15:
Thanks. I've just raised PR with this fix. FYI anyone can make PRs that include fixes like this - it's just that only the project maintainers can merge them.

meee1, 2019-05-24:
i can create a pull request. but the c# library will change alot. the one here was designed to work with the micro framework. where as mine was maintaining the c style structure. so what im trying to say is my implementation will break other of the existing generated code.

hamishwillee, 2019-05-27:
@meee1 As discussed offline, a PR to pull your version in, replacing the old version, would be much appreciated. People who want to use the old version can still do so, albeit with a few hoops to grab the code. Everyone else will appreciate a library that can do MAVLink 2 and 1.

I'll post a request on MAVLink channels to see if we can find anyone who is actually using the current CS implementation and notify them.

Can you give me a very approximate ETA for the PR?

PRIMETSS, 2019-11-07:
Just tried to generate MAVLink v2 classes and getting CRC errors parsing packets.
Came back to this post, and Im assuming MAVLink2 on CS is not doable?
Was working for MAVLink v1, but hit conflicts with message ID's already used, so thought I'd go to v2.
Anyone know of any C# MAVLink2 solutions?

meee1, 2019-11-07:
I have put in a pull request for this With mavlink 2 support.

PRIMETSS, 2019-11-07:
Thank-you for your efforts!!
I will find it tomorrow and give it a go.

meee1, 2019-11-07:
@PRIMETSS check nuget as well. ive packaged it on there as well

PRIMETSS, 2019-11-07:
Got it Talking to a old PIX OK. 
Looks good! Thankyou! 10 mins into it, but just wondering 1)Whats the log4net components. 2) Whats the best way to add custom Messages etc like you could using pymavlink XML files that then when and produced the CS classes. Just Hard code them in?

meee1, 2019-11-07:
Log4net see https://logging.apache.org/log4net/

I will post an example of custom message

meee1, 2019-11-07:
https://github.com/ArduPilot/MissionPlanner/tree/master/ExtLibs/MavlinkMessagePlugin

PRIMETSS, 2019-11-08:
Thanks Michael. Really appreciate the effort on this new version of the MAVLink CS library!! Its great! Now have MAVLinkv1 and v2 running, and was able to retro fit (aka re-write) the new into project and now able to inject MAVink messages and route them to multiple COM and UDP 'channels'
Wasn't aware pymavlink was inc. in ArduPilot Repo, and was also able to use the py gens to add the custom message types I wanted (not being absorbed by MissionPlanner, they are just injected and extracted on the ROV's Linux host and acted on outside of AP/MP)

Thankyou again for your efforts again!, I didn't know you were THE 'Micheal B'!!!

hamishwillee, 2019-11-10:
Cross link created for anyone looking at this: https://github.com/ArduPilot/pymavlink/pull/322

mavera3, 2019-11-27:
Hi, I am interested in trying out this new CS 2.0 library. How can I do so?

Thanks

meee1, 2019-11-27:
@mavera3 please try https://www.nuget.org/packages/MAVLink/

alexmarshall132, 2020-07-31:
> @mavera3 please try https://www.nuget.org/packages/MAVLink/

Thank you very much for this.

yakoverlich, 2020-12-02:
Hi Michael,
I'm C# programmer and we need to generate Mavlink V2 C#  code and some customer dialect also.
I'm new in Mavlink and Python filed, however, during my tries I succeed to generate Mavlink V 1.0 C# using mavgen.
This success occurred  due to well explained steps in following sites
https://mavlink.io/en/getting_started/installation.html
https://mavlink.io/en/getting_started/generate_libraries.html

I understand that you developed a way to generate Mavlink V2 C# common/customer dialect.
Can you please write a steps like in the mentioned sites to generate Mavlink v.2 C#
  Best Regards,
    Yakov

meee1, 2020-12-02:
@yakoverlich please check this https://github.com/ArduPilot/pymavlink/pull/322

yakoverlich, 2020-12-02:
Michael,
Excuse me to stupid questions.
I once more watched on  [#322](https://github.com/ArduPilot/pymavlink/pull/322) just now. 
Only one think that I see to do is to insert all difference that shown in [Generator CS update to support mavlink 2.0](https://github.com/ArduPilot/pymavlink/pull/322/commits/cb48f78686514e811de69356c60518a372f99770) :
 - insert all these files 
  MAVLink.csproj 
  MAVLinkMessage.cs 
  MavlinkCRC.cs 
  MavlinkParse.cs 
  MavlinkUtil.cs 
  
     into standard mavgen path:
     mavlink\pymavlink\generator\CS
   
- delete  folder mavlink\pymavlink\generator\CS\common

-  copy file 
  mavlink\pymavlink\dialects\v20\common.xml
to 
 mavlink\message_definitions\V20

Then run mavgen command

python -m pymavlink.tools.mavgen --lang=CS --wire-protocol=2.0 --output=generated/v2 message_definitions/v2.0/common.xml

meee1, 2020-12-02:
i personally use the files here
https://github.com/ArduPilot/MissionPlanner/tree/master/ExtLibs/Mavlink

python -m pymavlink.tools.mavgen --lang=CS --wire-protocol=2.0 "message_definitions\ardupilotmega.xml" "message_definitions\offspec.xml"

yakoverlich, 2020-12-02:
Michel,
I'm trying understand your last post.
1. Are correct all my steps for merge [Generator CS update to support mavlink 2.0](https://github.com/ArduPilot/pymavlink/pull/322/commits/cb48f78686514e811de69356c60518a372f99770) into mavgen tree?

2. You recommend me to use your command [python -m pymavlink.tools.mavgen ...] .
Let me understand it. According to mavgen help:
>python -m pymavlink.tools.mavgen -h
usage: mavgen.py [-h] [-o OUTPUT] [--lang {C,CS,JavaScript,TypeScript,Python,Lua,WLua,ObjC,Swift,Java,C++11}]
                 [--wire-protocol {0.9,1.0,2.0}] [--no-validate] [--error-limit ERROR_LIMIT] [--strict-units]
                 XML [XML ...]

 **ardupilotmega.xml** and  **offspec.xml** are dialects that mavgen will create C# sharp code from.
If it is correct, please help me to run my goal: create C# code with:
 - common.xml (standard) dialect
 - my private dialect my_dialect.xml

Due to I have no any idea regarding **ardupilotmega.xml** and  **offspec.xml**, I think to go by regular way described in [Mavlink ](https://mavlink.io/en/messages/common.html) site, to use common.html.

So, if I copy **common.xml** and **my_dialect.xml** into **mavlink\message_definitions** folder,
my command line should be:
 python -m pymavlink.tools.mavgen --lang=CS --wire-protocol=2.0 "message_definitions\common.xml" "message_definitions\my_dialect.xml"

If it is correct, I have only one question. Can I use file
mavlink\message_definitions\v1.0\common.xml
that is identical to
mavlink\pymavlink\dialects\v20\common.xml

Note: Following files  are different:
 mavlink\message_definitions\v1.0\common.xml
 ExtLibs\Mavlink\message_definitions\common.xml

Michael, sorry, I very confused what to do.

meee1, 2020-12-02:
ive just rebased the original pull request
https://github.com/ArduPilot/pymavlink/pull/322

i would suggest ensuring it works first
ie run
python -m pymavlink.tools.mavgen --lang=CS --wire-protocol=2.0 "message_definitions\ardupilotmega.xml"

and then change as you require. the path to your xml can be anything you need/want it to be

yakoverlich, 2020-12-03:
Michael,
I am extremely grateful your help.
I succeeded to create CS V2. 
  Best Regards,
    Yakov

auturgy, 2021-09-01:
C# generator has been updated https://github.com/ArduPilot/pymavlink/pull/322

closed 2021-09-01 by auturgy