ArduPilot/pymavlink #155: Failed Building wheel for lmxl error

AsternA, 2018-02-28:
Hey guys, Im a noob RPi user and an even bigger beginner at linux.
For some reason i can not get the pymavlink to install.
While connected to the RPi - according to 
http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html
everything works fine until the commands:

sudo pip install pymavlink
sudo pip install mavproxy

the scripts appears to be frozen on
Running setup.py bdist_wheel for lxml...
after too long of waiting i press ctrl+C and i get the errror

Failed building wheel for lxml.

this error did not happen when i had APsync (but had no com between pi and pixhawk) so im trying now on raspian.
any ideas?

Thanks!

khancyr, 2018-02-28:
hello,

Try sudo apt install python-lxml libxml2-dev libxslt-dev

AsternA, 2018-02-28:
Looks like it installed!!

sudo apt python-lxml solved it! thanks!!

closed 2018-02-28 by AsternA