ArduPilot/pymavlink #161: Java mission messages throw exception

jacobsenmd, 2018-03-14:
Several of the mission messages generated in Java throw IndexOutOfBoundsException when unpacked. So far I have seen this behavior with msg_mission_count, msg_mission_item, and msg_mission_ack.

For example, msg_mission_count.unpack():

```
public void unpack(MAVLinkPayload payload) {
        payload.resetIndex();
              
        this.count = payload.getUnsignedShort();
              
        this.target_system = payload.getUnsignedByte();
              
        this.target_component = payload.getUnsignedByte();
              
        this.mission_type = payload.getUnsignedByte();
    }
```

The final line throws the following exception (as recorded in Android adb):

```
11-22 16:32:23.321 10755-10882/io.diux.rogue.rosettadrone D/Rosetta: exception
                                                                     java.lang.IndexOutOfBoundsException: index=4 out of bounds (limit=4)
                                                                         at java.nio.Buffer.checkIndex(Buffer.java:528)
                                                                         at java.nio.HeapByteBuffer.get(HeapByteBuffer.java:118)
                                                                         at com.MAVLink.Messages.MAVLinkPayload.getUnsignedByte(MAVLinkPayload.java:62)
                                                                         at com.MAVLink.common.msg_mission_count.unpack(msg_mission_count.java:81)
                                                                         at com.MAVLink.common.msg_mission_count.<init>(msg_mission_count.java:100)
                                                                         at com.MAVLink.MAVLinkPacket.unpack(MAVLinkPacket.java:420)
```

WickedShell, 2018-03-14:
This isn't a relevant issue against ardupilot/pymavlink. This is due to changes incorrect changes in  mavlink/pymavlink. Sorry to be bouncing you back and forth between projects on this, I've refused to accept the proposed changes into the pymavlink due to a number of breakages likes this, and the original author having not revisted them.

closed 2018-03-14 by WickedShell

auturgy, 2018-03-14:
Background and details are here:
https://github.com/ArduPilot/pymavlink/pull/132
https://github.com/ArduPilot/pymavlink/pull/127