ArduPilot/pymavlink #172: Update pypi with DFReader changes

polastre, 2018-04-26:
Hi,

This is a request to update pypi with a new release that reflects the changes for Python3 to `DFReader.py` since the Jan 2018 release:
https://github.com/ArduPilot/pymavlink/commits/master/DFReader.py

Thank you.

peterbarker, 2018-04-27:
On Thu, 26 Apr 2018, Joe Polastre wrote:

> This is a request to update pypi with a new release that reflects the
> changes for Python3 to DFReader.py since the Jan 2018 release:
> https://github.com/ArduPilot/pymavlink/commits/master/DFReader.py

There are outstanding bugs in this code which affect Python2 users.  I 
have local hacks to work around them.

https://github.com/peterbarker/pymavlink/commit/557e99613e26da4f9c5d1dd492bf3f3c319b9deb

khancyr, 2018-10-04:
new release have come since April. Closing

closed 2018-10-04 by khancyr