ArduPilot/pymavlink #174: Home position change in local frame coordinate system

alexx600, 2018-04-30:
Is it possible to change home position according to local NED not to lattitude and longitude?

According to http://mavlink.org/messages/common#SET_HOME_POSITION “The global and local positions encode the position in the respective coordinate frames”. What shoud i send as a lattitude and longitute if i would like to move home positton in local frame( for exampe 2 m forwad)? I tried 0,0 as a lattituede and longitude but this caused reset home position

pietrodn, 2018-05-23:
You can surely generate the `SET_HOME_POSITION` message with pymavlink.
The problem is that the autopilot may have an incomplete implementation. For example, PX4 does not support setting the home position in local frame coordinates. You can work around this by converting local coordinates to global in client code. However, this would require to know the local GPS origin which is not readily available, but you can compute it, in turn, from the pair of corresponding global and local frames.

amilcarlucas, 2018-10-19:
This is not a pymavlink issue. This is specific to the flight controller FW (Ardupilot, PX4 or other implementations).
Closing it

closed 2018-10-19 by amilcarlucas