ArduPilot/pymavlink #191: More unit tests?

stephendade, 2018-06-18:
Is there interest in adding more unit tests to pymavlink? I've been attempting to use for some Python3 projects, but without much success.

Looking at the ``.travis.yml`` file, it appears that the unit testing is fairly light-on.

I've previously added a suite of unit tests to CanberraUAV's imaging project (https://github.com/CanberraUAV/cuav), so I thought I could do something similar here.

magicrub, 2018-06-18:
in mavlink/mavlink we discussed adding larger CI tests to mavlink. Specifically, making a real spec that GCSs must test against to become "compliant". It's just in brain-storming mode now, but I wonder if picture imaging like this could be part of it somehow.

peterbarker, 2018-06-19:
On Mon, 18 Jun 2018, Stephen wrote:

> Is there interest in adding more unit tests to pymavlink? I've been
> attempting to use for some Python3 projects, but without much success.

YES!  Unequivocally YES! :-)

magicrub, 2018-06-19:
https://github.com/mavlink/mavlink/issues/888#issuecomment-397014470

please attend!

stephendade, 2018-06-19:
@peterbarker - ok I'll start on improving the testing framework and add tests for a few of the python files as a starting point.

@magicrub - Thanks for the info! I'll definitely attend :)

khancyr, 2018-06-19:
Yes ! I can help testing this ! or correct python3

stephendade, 2018-06-21:
@khancyr - look at #192. The first unit tests are ready!