ArduPilot/pymavlink #220: Generated code comments omit units and enum names

hamishwillee, 2018-09-20:
Recently we updated the message format to move units and type (enum) information from the message descriptions into specific attributes. 

The generated comments (at least for python) need to be updated to reflect this - the information was removed from the description which is included in code comments.

FYI @amilcarlucas

amilcarlucas, 2018-09-20:
I fixed it for C and C++ code generation, but I did not update the comments generation for the other generated languages :(

hamishwillee, 2018-09-20:
@amilcarlucas YOu rock. Thanks! I might do it for Python while I'm playing with it.

amilcarlucas, 2018-10-22:
Yes, you are correct C, C++, and python are done but the other languages are still missing... reopening