ArduPilot/pymavlink #222: Pip installation fails

umer936, 2018-09-30:
Attempting to run `sudo pip2 install future pymavlink MAVProxy` fails while trying to install pymavlink==2.2.8. Happened yesterday on two machines. Odd as I have installed pymavlink==2.2.8 on ~20 machines in the exact same manner days before.

It fails saying, "No XML message definitions found." 

I searched as much as I could. Appears to be the exact same issue as https://github.com/ArduPilot/pymavlink/issues/41 and https://github.com/ArduPilot/pymavlink/issues/69; however, both claim to be fixed. This is on Ubuntu 18.04 after installing ROS Melodic. 

I ended up fixing this by `sudo pip2 install pymavlink==2.2.7`, which worked no problem. 

I know the following is a MAVProxy issue (and I plan on making a comment on the respective Issue there), but I have included the following for context. 
I then installed MAVProxy, ignoring dependancies so it wouldn't try to update pymavlink: `sudo pip2 install MAVProxy --no-deps`. 
To get things to run, I then had to modify mavproxy.py manually with PR: https://github.com/ArduPilot/MAVProxy/pull/544

pmtischler, 2018-09-30:
Our continuous integration tests caught this too. Started failing when version went from [2.2.10](https://travis-ci.org/auvsi-suas/interop/builds/434131002) to [2.2.11](https://travis-ci.org/auvsi-suas/interop/builds/435120131), and we've since [constrained the version to 2.2.10](https://github.com/auvsi-suas/interop/commit/95b2e72538d166d06cb954cbe26d010d6c0fbdb9).

umer936, 2018-09-30:
@pmtischler I don't think the issue should be limited to 2.2.10 as pip said it was trying to install 2.2.8

tridge, 2018-09-30:
I have done a new release 2.2.12 which should fix this

closed 2018-09-30 by tridge