ArduPilot/pymavlink #223: Stuck on pymavlink 2.06

benkawecki, 2018-10-03:
Hello,

Currently having an issue updating pymavlink on Raspbian. I was stuck on version 2.0.6 (not sure why that version was installed in the first place). I tried update to the newest version in order to satisfy requirements within dronekit but pip wouldn't let me. It gave me the following error:

pi@raspberrypi:~ $ sudo pip install --upgrade pymavlink
Requirement already up-to-date: pymavlink in /usr/local/lib/python2.7/dist-packages

I figured there's some requirement from another package that's causing this, but I couldn't pin down which one it is. Here is a list of my current <code> pip list</code>:

<details><summary>pip list output</summary>
<p>

automationhat (0.1.0)
Babel (2.6.0)
beautifulsoup4 (4.5.3)
blinker (1.3)
blinkt (0.1.1)
buttonshim (0.0.2)
Cap1xxx (0.1.3)
chardet (2.3.0)
click (6.6)
colorama (0.3.7)
cryptography (1.7.1)
cycler (0.10.0)
droneapi (1.5.0)
dronekit (2.9.1)
drumhat (0.1.0)
enum34 (1.1.6)
envirophat (1.0.0)
ExplorerHAT (0.4.2)
Flask (0.12.1)
fourletterphat (0.1.0)
functools32 (3.2.3.post2)
future (0.15.2)
gpiozero (1.4.1)
gps (3.9)
html5lib (0.999999999)
idna (2.2)
ipaddress (1.0.17)
iso8601 (0.1.12)
itsdangerous (0.24)
Jinja2 (2.8)
keyring (10.1)
keyrings.alt (1.3)
lxkeymap (0.1)
lxml (3.7.1)
MarkupSafe (0.23)
matplotlib (2.0.0)
MAVProxy (1.6.4)
mcpi (0.1.1)
microdotphat (0.2.1)
monotonic (1.2)
mote (0.0.3)
motephat (0.0.2)
numpy (1.12.1)
oauthlib (2.0.1)
pantilthat (0.0.5)
phatbeat (0.1.0)
pianohat (0.1.0)
picamera (1.13)
picraft (1.0)
piglow (1.2.4)
pigpio (1.38)
Pillow (4.0.0)
pip (9.0.1)
protobuf (3.6.0)
pyasn1 (0.1.9)
pycrypto (2.6.1)
pygame (1.9.3)
pygobject (3.22.0)
pyinotify (0.9.6)
PyJWT (1.4.2)
pymavlink (2.0.6)
pyOpenSSL (16.2.0)
pyparsing (2.1.10)
pyserial (3.2.1)
python-dateutil (2.7.3)
pytimeparse (1.1.8)
pytz (2016.7)
pyxdg (0.25)
rainbowhat (0.0.2)
requests (2.5.1)
requests-oauthlib (0.7.0)
RPi.GPIO (0.6.3)
RTIMULib (7.2.1)
scrollphat (0.0.7)
scrollphathd (1.2.1)
SecretStorage (2.3.1)
sense-emu (1.0)
sense-hat (2.2.0)
setuptools (33.1.1)
simplejson (3.10.0)
six (1.10.0)
skywriter (0.0.7)
sn3218 (1.2.7)
spidev (3.3)
subprocess32 (3.2.7)
touchphat (0.0.1)
twython (3.4.0)
unicornhathd (0.0.3)
urllib3 (1.19.1)
webencodings (0.5)
Werkzeug (0.11.15)
wheel (0.29.0)
wxPython (2.8.12.1)
wxPython-common (2.8.12.1)
zulu (0.12.1)
</p>
</details>

I then stupidly tried to uninstall and reinstall pymavlink to see what effect that would have. Now I can't install pymavlink and have the following error. 

[image]

I'm not sure where to go now, so if anyone has any ideas I would be appreciative. 

Thanks,

peterbarker, 2018-10-04:
That's odd.  I assume you've tried it a few times?  That could be a metadata issue; tridge did a release recently and bumped version numbers - it's possible something's Not Right.

benkawecki, 2018-10-04:
Peter,

You can go ahead and close the issue, it seems that I had an issue with my NTP timesync which was causing all sorts of problems with pip. This was the root cause of the issue. 

Thanks

khancyr, 2018-10-04:
Thanks for letting us know ! 
Closing

closed 2018-10-04 by khancyr