ArduPilot/pymavlink #245: mavparmdiff.py does not catch parameter differences

WickedShell, 2018-11-20:
Parameter files from `mavparms.py` have differences which `mavparmdiff.py` does not detect. Included a trimmed version of the files. The difference I'm tracking in the uploaded files is `ARSPD_OFFSET`

[parms.zip](https://github.com/ArduPilot/pymavlink/files/2598278/parms.zip)

peterbarker, 2018-11-26:
https://github.com/ArduPilot/pymavlink/blob/master/mavparm.py#L13 <- this is considered to be a volatile value and is explicitly excluded.

WickedShell, 2018-11-26:
That's pointlessly bad for a param comparison tool, as it breaks the main usage, detect differences. (In this case for example this offset is actually what I was looking for/expecting)

peterbarker, 2018-11-26:
On Sun, 25 Nov 2018, WickedShell wrote:

> That's pointlessly bad for a param comparison tool, as it breaks the main usage, detect differences. (In this case for example this offset is actually
> what I was looking for/expecting)

I guess the point is that they're *always* going to be different....

The same code is used within mavproxy, so a simple option to the 
mavparmdiff.py wouldn't be sufficient to change behaviour everywhere...

tridge, 2018-11-26:
I think we should have a command line option to not exclude the volatiles

WickedShell, 2018-11-26:
> I guess the point is that they're *always* going to be different....

Not actually true for `ARSPD_OFFSET` which is why I was looking for it (among others originally).

> I think we should have a command line option to not exclude the volatiles

That would work for me.

WickedShell, 2023-10-12:
This was done awhile ago as an opt in.

closed 2023-10-12 by WickedShell