ArduPilot/pymavlink #252: self.mav.target_component ： Automatically connect to the camera instead of the drone

xdwgood, 2018-12-28:
Hello all：
I use pymavlink to connect to a drone that downloaded px4 firmware.
My drone is equipped with a camera that uses the mavlink protocol.We define camera target_component in the mavlink protocol as 100.
I follow this ： https://www.ardusub.com/developers/pymavlink.html

`self.mav = mavutil.mavlink_connection('udpin:0.0.0.0:14550')`

```
self.mav.command_long_send(self.mav.target_system,self.mav.master.target_component,
    mavutil.mavlink.MAV_CMD_COMPONENT_ARM_DISARM,0,
    1, 0, 0, 0, 0, 0, 0)
```

I sent the arm command, but sometimes the drone did not receive it.I debugged it and found that the drone could not accept this command when `self.mav.master.target_component `was 100.The drone receives this command when `self.mav.master.target_component ` is 1.

So my question is how do I send commands to the drone instead of the camera. I know I can manually set it like this:
```
self.mav.command_long_send(1,1,
    mavutil.mavlink.MAV_CMD_COMPONENT_ARM_DISARM,0,
    1, 0, 0, 0, 0, 0, 0)
```
Is this correct? Or what better way do we have? Thank you :smiley:

xdwgood, 2019-01-03:
@tridge   Could you have a look? thanks :smiley:

amilcarlucas, 2019-01-06:
yes it is correct, use 100 for the camera and 1 for the drone