ArduPilot/pymavlink #259: C generated MAVLINK_MESSAGE_CRCS msg_len wrong for messages with extensions

julienazd, 2019-01-25:
When MAVLINK_MESSAGE_CRCS is generated by [mavgen_c.py](generator/mavgen_c.py#L111), the field msg_len is generated using the message_min_lengths. When dealing with a message which has extensions, if extensions fields have a value of zero, they are truncated. 
On the receiver side, in the function  [mavlink_helpers.h](generator/C/include_v2.0/mavlink_helpers.h#L743), we received the first bytes of the message, the extension is not send, and when we check the message length received against the msg_len expected, it looks like we received the whole message, because msg_len is the length without extensions, see `if (e && status->packet_idx < e->msg_len)`. Because we consider we received the full message length, we consider it has not been truncated, and so we don't execute what's inside the if, we don't pad the remaining bytes with zeros and we end up having a message payload with uninitialized data on the receiver side.
To fix this either the field msg_len must have the maximum length, with the extension, or we must have another length field with this maximum length.

khancyr, 2019-01-25:
the issue can be easily produce with upstream COMMAND_ACK, like this : 
change the COMMAND_ACK to  update version with extension
update mavproxy message definition, launch sitl
then : module load message
message COMMAND_ACK 11 1 22 33 44 55
At this point everything is fine.
launch again 
message COMMAND_ACK 11 1 22 33 44 0
as the mavlink buffer (rxmsg isn't clear) and we don't fill it with zero because of the length issue, then this command ack will be transmit to target_component 55.

khancyr, 2019-01-25:
ping @peterbarker @tridge

magicrub, 2019-01-25:
possibly related:
https://en.wikipedia.org/wiki/Heartbleed