ArduPilot/pymavlink #262: Lua/Wireshark: No module named 'pymavlink.tools'

guglie, 2019-01-30:
I'm trying to generate a Lua dissector for Wireshark.

Following the instructions on https://github.com/ArduPilot/pymavlink/blob/c18db673f0214b683bc330587bc11ce2818a8e8d/generator/mavgen_wlua.py#L9

I get this error
`ModuleNotFoundError: No module named 'pymavlink.tools'`

I know that a partial solution could be the one proposed at https://github.com/ArduPilot/pymavlink/issues/133 but how can I solve this issue?

guglie, 2019-02-06:
I made it work using the path to the mavgen module, but it feels like a workaround not the right solution...

E.g. (from the mavlink folder)
`python pymavlink/tools/mavgen.py  --lang=WLua message_definitions/v1.0/common.xml  -o /Applications/Wireshark.app//Contents/PlugIns/wireshark/mavlink.lua`

(The output folder here is the Wireshark plugin folder for Mac OSX)

pietrodn, 2019-02-21:
Hi, it looks like the `pymavlink` package is not in the `PYTHONPATH`.
How did you install the pymavlink module? Are you in a virtual environment? Which version of Python are you using?

vogt31337, 2023-03-09:
I just found out, that in the site-package folder (where pip installed the pymavlink package), the tools subfolder is missing. Seems during installation the subfolder tools went missing. But the pypi.org package contains the tools subfolder...