ArduPilot/pymavlink #294: mavnative/mavnative.c:334:13: error: conflicting types for '__assert_fail'

ingframin, 2019-03-24:
**When I try to install pymavlink from command line I get the following:**
running build_ext
building 'mavnative' extension
aarch64-openwrt-linux-musl-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/builder/aarch64_cortex-a53/build/sdk/build_dir/target-aarch64_cortex-a53_musl/Python-3.6.5:Python-3.6.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/builder/aarch64_cortex-a53/build/sdk/build_dir/target-aarch64_cortex-a53_musl/Python-3.6.5:Python-3.6.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/builder/aarch64_cortex-a53/build/sdk/build_dir/target-aarch64_cortex-a53_musl/Python-3.6.5:Python-3.6.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/builder/aarch64_cortex-a53/build/sdk/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/builder/aarch64_cortex-a53/build/sdk/staging_dir/target-aarch64_cortex-a53_musl/include -I/builder/aarch64_cortex-a53/build/sdk/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.3.0_musl/usr/include -I/builder/aarch64_cortex-a53/build/sdk/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.3.0_musl/include/fortify -I/builder/aarch64_cortex-a53/build/sdk/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.3.0_musl/include -fPIC -Igenerator/C/include_v1.0 -Igenerator/C/include_v2.0 -Imavnative -I/usr/include/python3.6 -c mavnative/mavnative.c -o build/temp.linux-aarch64-3.6/mavnative/mavnative.o
mavnative/mavnative.c:334:13: error: conflicting types for '__assert_fail'
 extern void __assert_fail(const char *__assertion, const char *__file, unsigned int __line, const char *__function)
             ^
In file included from mavnative/mavnative.c:14:0:
/usr/include/assert.h:19:16: note: previous declaration of '__assert_fail' was here
 _Noreturn void __assert_fail (const char *, const char *, int, const char *);
                ^
mavnative/mavnative.c: In function 'msg_to_py':
mavnative/mavnative.c:659:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(fnum = 0; fnum < info->num_fields && objValid; fnum++) {
                        ^
error: command 'aarch64-openwrt-linux-musl-gcc' failed with exit status 1
-------------------------------------------------------------------------------------------------------
It seems that the function **_extern void __assert_fail(const char *__assertion, const char *__file, unsigned int __line, const char *__function)_** is declared differently than in assert.h or Python.h. 

I am using:
- gcc - 5.4.0-3
- Python 3.6.5-4
- OpenWRT running on a RaspberryPi 3B

romejoe, 2019-07-09:
We just ran into this issue too while trying to build in Alpine 3.10. It appears that pymavlink is intended to compile against glibc, not musl. We were able to compile and install it by patching 

`extern void __assert_fail(const char *__assertion, const char *__file, unsigned int __line, const char *__function)`
 to
 `extern void __assert_fail(const char *__assertion, const char *__file, int __line, const char *__function)` 
in `mavnative/mavnative.c`. However, we had to switch to glibc before we could test things with musl, so I don't know if there would be any issues at runtime.

romejoe, 2019-07-09:
Replacing that line with
```
#ifdef __GLIBC__
extern void __assert_fail(const char *__assertion, const char *__file, unsigned int __line, const char *__function)
#else
extern void __assert_fail(const char *__assertion, const char *__file, int __line, const char *__function)
#endif
```
should let it compile with both libraries, but I haven't tested this at all.

khancyr, 2019-07-09:
@peterbarker @tridge What about to make mavnative optional on pymavlink install ? I doubt it is really used. Without it we could largely simplify pymavlink install !(and make pymavlink test much more greener!)

khancyr, 2019-07-09:
@romejoe hello, could you open a PR with your solution, it seems good to me.

romejoe, 2019-07-10:
@khancyr just opened a pull request(#337 ) to address this with the above fix.