ArduPilot/pymavlink #301: Ubuntu error

cathwYang, 2019-04-16:


cathwYang, 2019-04-16:
In the compilation process，the last step ,when i do "make px4-v2",it reports "error:No rule to make target".I tried many solutions,but no work.I followed the steps on the website（github）.Please,could someone help me？I am a Green hand， Can big guys help me?If someone has met the same problem and solved it, can you tell me your solution? Thank you very much！

auturgy, 2019-04-16:
Your question doesn’t seem to relate to pymavlink.
For instructions on building ArduPilot in Ubuntu, see https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md
The particular build target you’re using has been deprecated.
If you’re trying to build PX4, there’s instructions on their website.

closed 2019-04-16 by auturgy

cathwYang, 2019-04-16:
> _No description provided._

> Your question doesn’t seem to relate to pymavlink.
> For instructions on building ArduPilot in Ubuntu, see https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md
> The particular build target you’re using has been deprecated.
> If you’re trying to build PX4, there’s instructions on their website.
Thank you for answering my question, but I still have some doubts. I compiled ardupilot according to a tutorial. Finally, I should generate such a file as the following.
[image]

auturgy, 2019-04-16:
Best to take this to discuss.ardupilot.org and continue there.
This issues list is for pymavlink issues.