ArduPilot/pymavlink #308: How to use it with the data as"FE XX XX ... XX"

wangkailovebaojiakun, 2019-04-25:
I have some data as"FE 1E 6D 01 01 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F 27 00 00 00 00 00 00 00 00 3F 3E FE 1C 65 01 01 21 C2 D6 12 00 00 00 00 00 00 00 00 00 F6 04 00 00 F6 04 00 00 FF FF 03 00 00 00 44 6D 44 9D" how to deal it with pymavlink?

peterbarker, 2019-04-25:
On Thu, 25 Apr 2019, wangkailovebaojiakun wrote:

> I have some data as"FE 1E 6D 01 01 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F 27 00 00 00 00 00 00 00 00 3F 3E FE 1C 65 01 01 21 C2
> D6 12 00 00 00 00 00 00 00 00 00 F6 04 00 00 F6 04 00 00 FF FF 03 00 00 00 44 6D 44 9D" how to deal it with pymavlink?

https://github.com/ardupilot/ardupilot/blob/master/Tools/scripts/uploader.py#L214

wangkailovebaojiakun, 2019-04-25:
> On Thu, 25 Apr 2019, wangkailovebaojiakun wrote: I have some data as"FE 1E 6D 01 01 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F 27 00 00 00 00 00 00 00 00 3F 3E FE 1C 65 01 01 [...]

sorry，I mean how to resolving or decode it

peterbarker, 2019-04-25:
On Thu, 25 Apr 2019, wangkailovebaojiakun wrote:

> On Thu, 25 Apr 2019, wangkailovebaojiakun wrote: I have some data as"FE 1E 6D 01 01 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
>       00 00 0F 27 00 00 00 00 00 00 00 00 3F 3E FE 1C 65 [...]

It's an encoded mavlink packet.  You should be able to create a mavlink 
parser and feed the bytes (e.g. 0xFE as the first one) into the mavlink 
parser to decode the packet.  It should eventually indicate it's found the 
packet... and then you have your packet to play with...

amilcarlucas, 2020-09-24:
solved by @peterbarker above

closed 2020-09-24 by amilcarlucas