ArduPilot/pymavlink #309: No such file or directory:message_definitions/v1.0/ardupilotmega.xml

roseyangyu, 2019-04-29:
ERROR:dronekit.mavlink:Exception while receiving message: 
Traceback (most recent call last):
  File "dronekit-python/dronekit/mavlink.py", line 218, in mavlink_thread_in
  File "pymavlink-2.3.4-py2.7-linux-aarch64.egg/pymavlink/mavutil.py", line 1061, in recv_msg
  File "pymavlink-2.3.4-py2.7-linux-aarch64.egg/pymavlink/mavutil.py", line 267, in auto_mavlink_version
  File "pymavlink-2.3.4-py2.7-linux-aarch64.egg/pymavlink/mavutil.py", line 101, in set_dialect
  File "pymavlink-2.3.4-py2.7-linux-aarch64.egg/pymavlink/generator/mavgen.py", line 241, in mavgen_python_dialect
  File "pymavlink-2.3.4-py2.7-linux-aarch64.egg/pymavlink/generator/mavgen.py", line 136, in mavgen
  File "pymavlink-2.3.4-py2.7-linux-aarch64.egg/pymavlink/generator/mavgen.py", line 105, in mavgen_validate
IOError: [Errno 2] No such file or directory: '../../../../tmp/_MEIzZzD6G/message_definitions/v1.0/ardupilotmega.xml'

roseyangyu, 2019-04-29:
I have installed pymavlink==2.3.4 by "pip install pymavlink==2.3.4"

auturgy, 2019-04-29:
Try sudo pip install -U pymavlink==2.3.4

roseyangyu, 2019-04-29:
> Try sudo pip install -U pymavlink==2.3.4
> […](#)

I have tried.Same erros

roseyangyu, 2019-04-29:
Use "sudo MDEF=PATH_TO_message_definitions pip2 install . -vsudo MDEF=PATH_TO_message_definitions pip2 install . -v",it has error as "No XML message definitions found"

Huibean, 2019-05-08:
define MDEF with the right directory

roseyangyu, 2019-05-16:
Mavlink1 has not error.But mavlink2 still doesn't solve

amilcarlucas, 2020-09-24:
Retest it with pymavlink 2.4.11 and reopen if it still does not work

closed 2020-09-24 by amilcarlucas