ArduPilot/pymavlink #319: Problem with installation using setup.py

kaangoksal, 2019-05-17:
I cannot install pymavlink using the supplied setup.py file on ubuntu 18.04 for a virtualenv. The installation stops with a message "No XML message definitions found". Obviously it cannot find the message definition files. Where should I find the message definitions? The current pip package is outdated and does not include the recent changes on master.

kaangoksal, 2019-05-17:
For anyone else that is trying to install manually, I copied the message definitions from to pip repository and added it to my own fork.

WileCyot, 2019-08-30:
Do you have a link?

khancyr, 2019-08-30:
Mavlink definitions are on mavlink repository

bobpaul, 2019-09-11:
If the mavlink definitions are in a different repository and are required to build this repository, then the other repository should probably be a sub module of this one. Then we could do the standard `git submodules update` process.

khancyr, 2019-09-11:
Pymavlink is a submodules of Mavlink already. 
Mavlink dialect don't belong to pymavlink so we cannot includes them. 
But I agree that we shouldn't failed the installation if no dialect it provide

acxz, 2020-03-19:
I am also encountering the same issue.