ArduPilot/pymavlink #340: How to get 'LOCAL_POSITION_NED' parameter from uav ?

wuxianshen, 2019-07-19:
I try to make a offboard control of PX4 with pymavlink, and in the program I need to read 'LOCAL_POSITION_NED' message from PX4, but the results tend to be fixed as the first reading, they don't change at all.
Is there any problem with my implementation? Or is it because PX4 support not perfect?
`print([self.the_connection.messages['LOCAL_POSITION_NED'].x,
               self.the_connection.messages['LOCAL_POSITION_NED'].y,
               self.the_connection.messages['LOCAL_POSITION_NED'].z])`

Thank you!!

wuxianshen, 2019-07-20:
I call the request like this, but is this message deprecated?
`self.the_connection.mav.request_data_stream_send(self.the_connection.target_system,
                                                         self.the_connection.target_component,
                                                         # mavutil.mavlink.MAV_DATA_STREAM_POSITION, 5, 1)
                                                         mavutil.mavlink.MAV_DATA_STREAM_ALL, 5, 1)`

wuxianshen, 2019-11-21:
If we want to acquire new information from uav, we should call 
self.the_connection.recv_match(type='LOCAL_POSITION_NED', blocking=True)
to update the stored state.

closed 2019-11-21 by wuxianshen

rmackay9, 2019-11-21:
Assuming you're using ArduPilot, we've recently added [this wiki page](We've recently created this) with details of the various methods of getting the UAV to send messages.