ArduPilot/pymavlink #372: Serial connection has HEARTBEAT but no other message

yuchnw, 2019-12-02:
I am trying to connect pixracer to a raspberry pi 3 using serial connection.

> from pymavlink import mavutil
> the_connection = mavutil.mavlink_connection('/dev/ttyAMA0',baud=57600)
> the_connection.wait_heartbeat()
> print("Heartbeat from system (system %u component %u)" % (the_connection.target_system, the_connection.target_system))

Above is the code for establishing the connection and I successfully received the heartbeat printed as `Heartbeat from system (system 1 component 1)`
Then I tried to receive message from the pixhawk using
> msg = the_connection.recv_match(type='PING',blocking=True)
> if not msg:
>         print("no msg")
> if msg.get_type() == "BAD_DATA":
>     if mavutil.all_printable(msg.data):
>         sys.stdout.write(msg.data)
>         sys.stdout.flush()
> else:
>     print(msg)

When the message type is **SYS_STATU**, it printed out `SYS_STATUS {onboard_control_sensors_present : 917551, onboard_control_sensors_enabled : 2097167, onboard_control_sensors_health : 917551, load : 363, voltage_battery : 11443, current_battery : 0, battery_remaining : 100, drop_rate_comm : 0, errors_comm : 0, errors_count1 : 0, errors_count2 : 0, errors_count3 : 0, errors_count4 : 0}`

When the message type is **PING**, the output is `PING {time_usec : 3064363492, seq : 3061, target_system : 0, target_component : 0}`

When I tried other message types like **RAW_IMU**, **PARAM_SET**, etc, there is nothing printed out nor any error message. Looks like it's always waiting to receive the message. Does anyone have any idea about this issue? Any help would be appreciated!!!

yuchnw, 2019-12-02:
@julianoes @rmackay9
Sorry to disturb you but this issue has been on for days and I really have no idea how to solve it :(

peterbarker, 2019-12-02:
On Mon, 2 Dec 2019, Yuchen Wang wrote:

> Above is the code for establishing the connection and I successfully received the heartbeat printed as Heartbeat from system (system 1 component 1)
> Then I tried to receive message from the pixhawk [...]

Have you requested streams or individual messages from the autopilot?

MAV_CMD_SET_MESSAGE_INTERVAL or REQUEST_DATA_STREAM

joker-smiling, 2019-12-06:
I have met the same problem... But if use the command 'mavproxy.py --master=/dev/ttyAMA0 --baudrate = 57600' first, and use CTRL+C to stop it.Then, I use the scripts similar to yours and I could only get two kinds of messages "ALITITUDE，GLOBAL_POSITION_INT".

joker-smiling, 2019-12-06:
@peterbarker Hi , you talk about the ‘MAV_CMD_SET_MESSAGE_INTERVAL or REQUEST_DATA_STREAM’,when and where should I use it?

joker-smiling, 2019-12-06:
@peterbarker I used the 'REQUEST_DATA_STREAM', but it still can only get the heartbeat.....

peterbarker, 2019-12-06:
On Thu, 5 Dec 2019, yepzhang wrote:

> @peterbarker I used the 'REQUEST_DATA_STREAM', but it still can only get the heartbeat.....

Which autopilot are you targetting?

Please paste in a copy of the packet you're assembling.

Suggest you try a simple message time TIMESYNC to ensure you're actually 
getting messages to the vehicle correctly.

joker-smiling, 2019-12-06:
@peterbarker the hardware is pixhwak, and the firmware is ardupilot.

joker-smiling, 2019-12-06:
@peterbarker it really puzzles me. if I use the command 'mavproxy.py --master=/dev/ttyAMA0 --baud = 57600' first,then I stop it using 'CTRL+C' .finally I run the previous code(get all message), it show that I can only get two kinds message 'ALTITUDE' and 'GLOBAL_POSITION_INT'.

peterbarker, 2019-12-06:
On Fri, 6 Dec 2019, yepzhang wrote:

> @peterbarker it really puzzles me. if I use the command 'mavproxy.py --master=/dev/ttyAMA0 --baud = 57600' first,then I stop it using 'CTRL+C' .finally I
> run the previous code(get all message), it show that I can only get two kinds message 'ALTITUDE' and 'GLOBAL_POSITION_INT'.

When you run "status" in MAVProxy, does it show *lots* of different 
messages?

joker-smiling, 2019-12-06:
@peterbarker it shows this.
83: ATTITUDE {time_boot_ms : 176394, roll : -0.00454974919558, pitch : -0.0371655188501, yaw : 0.938688099384, rollspeed : 0.00066411937587, pitchspeed : -0.000120494456496, yawspeed : 4.98715307913e-05}
2: COMMAND_ACK {command : 400, result : 0}
83: GLOBAL_POSITION_INT {time_boot_ms : 176394, lat : 0, lon : 0, alt : 130, relative_alt : 130, vx : 0, vy : 0, vz : 1, hdg : 5378}
28: HEARTBEAT {type : 2, autopilot : 3, base_mode : 209, custom_mode : 2, system_status : 3, mavlink_version : 3}
778: PARAM_VALUE {param_id : GND_ABS_PRESS, param_value : 97867.7109375, param_type : 9, param_count : 775, param_index : 65535}
4: STATUSTEXT {severity : 6, text : PX4v2 00420029 33355110 35303335}

joker-smiling, 2019-12-09:
@peterbarker I will appreciate it that if you could tell me how to get the GPS,IMU and other messages.

peterbarker, 2019-12-09:
On Sun, 8 Dec 2019, yepzhang wrote:

> @peterbarker I will appreciate it that if you could tell me how to get the GPS,IMU and other messages.

I think you really need to start by working out why MAVProxy doesn't work 
correctly.

I'd be curious to see a mav.tlog from MAVProxy for this.

joker-smiling, 2019-12-10:
'WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk'
when I start the mavproxy, it shows that. But I can still set the command to the pixhwak and control it.

joker-smiling, 2019-12-10:
[mav.zip](https://github.com/ArduPilot/pymavlink/files/3942491/mav.zip)

joker-smiling, 2019-12-10:
I use this command 'sudo systemctl disable ModemManager.service' to disable the ModemManager, but it still gets the warning.

joker-smiling, 2019-12-10:
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect /dev/ttyACM1 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/ttyACM1
 MAV> online system 4
LAND> Mode LAND
`-APM: APM:Copter V3.6-dev (db7c8439)
APM: PX4: f84ee497 NuttX: c626db4d
APM: Frame: QUAD
APM: PX4v2 00420029 33355110 35303335

LAND> 
LAND> Received 775 parameters
Saved 775 parameters to mav_4_1.parm

LAND> 
LAND>

peterbarker, 2019-12-10:
On Mon, 9 Dec 2019, yepzhang wrote:

> mav.zip

That was a .rar, in fact :-)

Please try running MAVProxy.py with "--mav10" and see what happens. 
You're running a 2.5 year old version of ArduPilot on that board, which 
isn't great - it should still work, 'though!

joker-smiling, 2019-12-10:
LAND> status
LAND> Counters: Slave:0 MasterIn:[1074] FGearOut:0 FGearIn:0 MasterOut:43 
MAV Errors: 0
None
124: ATTITUDE {time_boot_ms : 78243, roll : 0.0405478253961, pitch : 0.0180831830949, yaw : -2.1027469635, rollspeed : 0.000648572051432, pitchspeed : -0.00111488054972, yawspeed : 0.000301446852973}
124: GLOBAL_POSITION_INT {time_boot_ms : 78243, lat : 0, lon : 0, alt : 270, relative_alt : 270, vx : 0, vy : -2, vz : 5, hdg : 23953}
45: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 9, system_status : 3, mavlink_version : 3}
777: PARAM_VALUE {param_id : STAT_RUNTIME, param_value : 613377.0, param_type : 6, param_count : 775, param_index : 65535}
4: STATUSTEXT {severity : 6, text : PX4v2 00420029 33355110 35303335}

peterbarker, 2019-12-10:
On Mon, 9 Dec 2019, yepzhang wrote:

Was this with --mav10?

joker-smiling, 2019-12-10:
yes

joker-smiling, 2019-12-14:
@peterbarker thank you! It finally solved. My firmware version is too low.

aabs7, 2020-08-09:
Same problem happened to me. I reinstalled Pymavlink at it's working fine.

aabs7, 2020-08-15:
Actually, Reinstalling pymavlink was not the fix. What I did was, set the stream rate parameters of sr1 (using mission planner or QGC) and rebooted the sitl.

kyrlon, 2021-03-01:
Not sure how different it is from interacting with USB, but I set the stream for all information before parsing for the messages i wanted:

```
the_connection = mavutil.mavlink_connection("/dev/ttyACM0", baud=115200)
# Request all data streams
data_rate = 1 # Hz
the_connection.mav.request_data_stream_send(the_connection.target_system, the_connection.target_component,
                                        mavutil.mavlink.MAV_DATA_STREAM_ALL, data_rate, 1)
```

csz5024, 2021-08-19:
> Not sure how different it is from interacting with USB, but I set the stream for all information before parsing for the messages i wanted:
>
> ```
> the_connection = [...]

This worked for me on Pixhawk 4 and Ardupilot, thanks!

auturgy, 2021-09-01:
Closed as resolved

closed 2021-09-01 by auturgy

dronesmedellin, 2022-04-01:
> Not sure how different it is from interacting with USB, but I set the stream for all information before parsing for the messages i wanted:
>
> ```
> the_connection = [...]

OMG it worked for me.... THANKS!!
Setup ==> Pixhawk 2 (Orange Cube) ---> Windows 10 ---> Connecting the usb port of the PC directly to the Pix usb port

`from pymavlink import mavutil
import time

# Start a connection listening to a UDP port
the_connection = mavutil.mavlink_connection("COM6", baud=115200)

# Wait for the first heartbeat 
#   This sets the system and component ID of remote system for the link
the_connection.wait_heartbeat()
print("Heartbeat from system (system %u component %u)" % (the_connection.target_system, the_connection.target_component))

# Request all data streams
data_rate = 120 # Hz
the_connection.mav.request_data_stream_send(the_connection.target_system, the_connection.target_component,
                                        mavutil.mavlink.MAV_DATA_STREAM_ALL, data_rate, 1)
# Once connected, use 'the_connection' to get and send messages
while True:
    msg = the_connection.recv_match(blocking=True)
    print(msg)
    `