ArduPilot/pymavlink #378: Norm of fft

tscao, 2020-01-08:
Hi,

In the mavfft script: 
https://github.com/ArduPilot/pymavlink/blob/f3af3255a48e34031bc5c0d89cdb806861d83b06/tools/mavfft_isb.py#L127

Why don't we multiply `d_fft` by `thing_to_plot.sample_rate_hz / len(d)` ? Like that, the units of the FFT outputs would have a physical meaning (m/s/s or rad/s).

peterbarker, 2020-05-07:
@andyp1per would you like to weigh in here?

andyp1per, 2020-05-07:
@tscao it's not amplitudes of the raw signals - it's power spectral density, the units are given on the axis. The important thing is that this is /Hz so that the values are comparable between graphs.

andyp1per, 2020-05-07:
It's this line that does it:
```
            # normalize output to averaged PSD
            psd = 2 * (sum_fft[sensor][axis] / counts[sensor]) / (sample_rates[sensor] * S2[sensor])
```
See https://holometer.fnal.gov/GH_FFT.pdf for details