ArduPilot/pymavlink #382: pymavlink install on Raspbian Buster fails

7bpm, 2020-02-01:
All dependencies have been installed successfully but `sudo pip3 install pymavlink --user` fails.

It looks like there's been a change over the last days or so, as a week ago I was able to install it with the exact same manner. 

The error I get is the following:

> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pymavlink
  Downloading https://files.pythonhosted.org/packages/b1/b4/37066db27b83f72fdb8e3fa988ec0ba901c6e39c9ac68018085af90863ec/pymavlink-2.4.5.tar.gz (3.4MB)
    100% |████████████████████████████████| 3.4MB 131kB/s
Collecting future (from pymavlink)
  Downloading https://www.piwheels.org/simple/future/future-0.18.2-py3-none-any.whl (491kB)
    100% |████████████████████████████████| 491kB 657kB/s
Collecting lxml (from pymavlink)
  Downloading https://files.pythonhosted.org/packages/39/2b/0a66d5436f237aff76b91e68b4d8c041d145ad0a2cdeefe2c42f76ba2857/lxml-4.5.0.tar.gz (4.5MB)
    100% |████████████████████████████████| 4.5MB 99kB/s
Building wheels for collected packages: pymavlink, lxml
  Running setup.py bdist_wheel for pymavlink ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-w70i_y3q/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-_rzweiuf --python-tag cp37:
  running bdist_wheel
  running build
  running build_py
  No XML message definitions found

7bpm, 2020-02-01:
Update...

I've noticed that pymavlink/setup.py has been changed 17 hours ago, as it is being prepared for the release of 2.4.5.

So, I was able to install pymavlink with: `sudo pip3 install pymavlink==2.4.4`

khancyr, 2020-02-01:
Hello,

Thanks for the report, the release got an error. We will fix it soon !

khancyr, 2020-02-01:
The error is that the pip package miss the message_definitions with the xml .

7bpm, 2020-02-01:
@khancyr 

Thank you for the prompt reply..!

Just let us know when it's fixed...

khancyr, 2020-02-01:
@7bpm it should be good now.

Please reopen the issue if it doesn't work.

closed 2020-02-01 by khancyr

7bpm, 2020-02-01:
just for the history of it...

I now get..:

> Successfully installed lxml-4.5.0 pymavlink-2.4.6

@khancyr thanks for sorting this out, so promptly...!!!

wx4cb, 2020-03-29:
trying to install from the GH repo - fresh install of rasbian I get the following

```
charlesb@TalonFC:/usr/src/mavlink/pymavlink $ python2 setup.py install
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
running install
running bdist_egg
running egg_info
creating pymavlink.egg-info
writing requirements to pymavlink.egg-info/requires.txt
writing pymavlink.egg-info/PKG-INFO
writing top-level names to pymavlink.egg-info/top_level.txt
writing dependency_links to pymavlink.egg-info/dependency_links.txt
writing manifest file 'pymavlink.egg-info/SOURCES.txt'
reading manifest file 'pymavlink.egg-info/SOURCES.txt'
writing manifest file 'pymavlink.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
No XML message definitions found

```

When trying to install under python3 using pip3 i get the following:
```
charlesb@TalonFC:/usr/src/mavlink/pymavlink $ pip3 install mymavlink --user
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mymavlink
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/mymavlink/
```

and under a manual install i get the following

```
charlesb@TalonFC:/usr/src/mavlink/pymavlink $ python3 setup.py install
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
running install
running bdist_egg
running egg_info
writing pymavlink.egg-info/PKG-INFO
writing dependency_links to pymavlink.egg-info/dependency_links.txt
writing requirements to pymavlink.egg-info/requires.txt
writing top-level names to pymavlink.egg-info/top_level.txt
reading manifest file 'pymavlink.egg-info/SOURCES.txt'
writing manifest file 'pymavlink.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
No XML message definitions found

```