ArduPilot/pymavlink #389: Sending OBSTACLE_DISTANCE message

bigboy061293, 2020-03-04:
Hello, 

I could do send DISTANCE_SENSOR by using master.mav.distance_sensor_send() but could not do the same for master.mav.obstacle_distance_send().

Could anyone show me the way? Thanks

bigboy061293, 2020-03-04:
I am using python by the way. By diving into the structure, I guess the message must be declared in right dialects (my case: ardupilotmege). Does anyone know the way to do?

closed 2020-03-10 by bigboy061293

jimmyfraiture, 2020-03-23:
Did you find a solution?

bigboy061293, 2020-03-24:
> Did you find a solution?

I did, the message was included already in the code, we don't need to add it.

jimmyfraiture, 2020-03-24:
In another version of mavlink then? Because when Install with pip3 I don't find it

bigboy061293, 2020-04-04:
> In another version of mavlink then? Because when Install with pip3 I don't find it

Sorry for this late,

I am using the pymavlink with python 2.7. As far as I know, pymavlink works well at this version of python. By following this post, it helps a lots:

https://www.ardusub.com/developers/pymavlink.html

But remember to add the dialect to your initialization :

`master = mavutil.mavlink_connection(UDP_IP_650, dialect = "ardupilotmega")`

jimmyfraiture, 2020-04-09:
I did exactly what they do and python still tell me 'MAVLink' object has no attribute 'obstacle_distance_send'.

Can I see what your code look like?

bigboy061293, 2020-04-28:
> I did exactly what they do and python still tell me 'MAVLink' object has no attribute 'obstacle_distance_send'.
> 
> Can I see what your code look like?

Sure and sorry for this late!
```

from pymavlink import mavutil
from pymavlink.dialects.v10 import ardupilotmega as mavlink1
from pymavlink.dialects.v20 import ardupilotmega as mavlink2
master = mavutil.mavlink_connection(UDP_IP_650_TCP, dialect = "ardupilotmega")

msg = None
    while not msg:
        master.mav.ping_send(
            time.time(), # Unix time
            0, # Ping number
            0, # Request ping of all systems
            0 # Request ping of all components
        )
        msg = master.recv_match()
        #print msg
        time.sleep(0.5)
#declare abc here
msss = mavlink2.MAVLink_obstacle_distance_message(
                    0, #time
                    3, #sensor type
                    abc, # abc is usually the array of 72 elements
                    1, #angular width
                    10, #min distance
                    maxDist, #max distance
                    1.53,
                    -55.08,
                    12
                    )
        time.sleep(0.01)
        master.mav.send(msss)
```