ArduPilot/pymavlink #390: mavlink_finalize_message_buffer prepares whole mavlink_message_t structure to be send.

rligocki, 2020-03-04:
Hey guys, 

during mavlink dev call I address this problem and we decided, that first I should write it down here so you can better understand it so we can find a solution. 

The issue is, that mavlink_finelize_message_buffer sends whole mavlink_message_t structure. Right now it assumes, that this structure has only one optional part (signature). After I add additional parts (mac and nonce) to make mavlink encryption possible, I found the problem, that in https://github.com/mavlink/qgroundcontrol/blob/95d2b14793fe9078942a8e7e536b36cb30da5c25/src/FactSystem/ParameterManager.cc#L529 function, the whole structure is sent with only length limitation. If length is bigger than non-optional part (header + payload + CRC) of the message it optional parts in order as they are defined in mavlink_types.h file, so there is no way to send nonce and mac without signature part. 

For now, I see the only solution in using _mav_finalize_message_chan_send function instead of mavlink_finalize_message_buffer function. 

The current version of _mavlink_message_t:

```
MAVPACKED(
typedef struct __mavlink_message {
	uint16_t checksum;     
	uint8_t magic;         
	uint8_t len;           
	uint8_t incompat_flags; 
	uint8_t compat_flags;  
	uint8_t seq;         
	uint8_t sysid;          
	uint8_t compid;         
	uint32_t msgid:24;     
	uint64_t payload64[(MAVLINK_MAX_PAYLOAD_LEN+MAVLINK_NUM_CHECKSUM_BYTES+7)/8];
	uint8_t ck[2];        
        uint8_t signature[MAVLINK_SIGNATURE_BLOCK_LEN]; <<<<< If the length of a message is set to bigger value then the length of mandatory parts, a signature is sent. That's Ok. 
}) mavlink_message_t;
```

My version of _mavlink_message_t: 

```
MAVPACKED(
typedef struct __mavlink_message {
	uint16_t checksum;     
	uint8_t magic;        
	uint8_t len;      
	uint8_t incompat_flags; 
	uint8_t compat_flags;   
	uint8_t seq;            
	uint8_t sysid;       
	uint8_t compid;        
	uint32_t msgid:24;      
	uint64_t payload64[(MAVLINK_MAX_PAYLOAD_LEN+MAVLINK_NUM_CHECKSUM_BYTES+7)/8];
	uint8_t ck[2];
    uint8_t signature[MAVLINK_SIGNATURE_BLOCK_LEN];   <<<< In this case, if I want to send mac and nonce, signature will be sent also. Even if I don't want to send signature.   
    uint8_t mac[16];
    uint8_t nonce[24];
}) mavlink_message_t;
```

Does it make sense or should I add even more information?

julianoes, 2020-03-04:
Could you change `mavlink_finalize_message_buffer` to work with what you're suggesting?

rligocki, 2020-03-04:
Problem is not exactly in mavlink_finalize_message_buffer. I changed it to add encryption capability, but for example, in qgroundcontrol (https://github.com/mavlink/qgroundcontrol/blob/95d2b14793fe9078942a8e7e536b36cb30da5c25/src/FactSystem/ParameterManager.cc#L529) this function sends the whole message with length limitation. Order is most important. It doesn't send optional parts based on incompat flag.

julianoes, 2020-03-05:
@rligocki I'm still confused. As part of your change you will have to change all the functions which assemble the bytes on the wire, including `mavlink_finalize_message_buffer`. Is that possible? And if not, can you explain to me why not?

rligocki, 2020-03-08:
Ok. Finally, I found out how it works. A bug has been fixed and now all messages between QGC and PX4 firmware are encrypted. Things like parameters or missions are also encrypted. But still, I need to check if FTP and Console work the same way.

julianoes, 2020-03-09:
So we can close this?

closed 2020-03-09 by rligocki