ArduPilot/pymavlink #397: OBSTACLE_DISTANCE message

jimmyfraiture, 2020-03-23:
Hello,
I'm trying to send OBSTACLE_DISTANCE message from a RPi to a PX4 using pymavlink but it doesn't seem to be implemented. Is it an alternative for that?
Is it a way to send 'brut' message or do I have to build the packet myself? 

Thanks

jmachuca77, 2020-03-23:
How did you install pymavlink? you might need to generate from source

jimmyfraiture, 2020-03-23:
Simply using pip3 like in the doc.
How should I do then?

jmachuca77, 2020-03-23:
Follow the instructions in this repo and install for developers 
https://github.com/ArduPilot/pymavlink#for-developers
use the latest mavlink definitions.

jimmyfraiture, 2020-04-09:
I did that but nothing changed

amilcarlucas, 2020-09-24:
That message is Work-in-progress. And as long as it remains so it will not be supported by the pymavlink release.
You can however use a locally compiled mavlink with the latest upstream mavlin/mavlink repository. That will work.

closed 2020-09-24 by amilcarlucas