ArduPilot/pymavlink #404: Inconsistent usage of mavlink_channel_t

len0rd, 2020-04-06:
This regards the generated C/C++ code.

`mavlink_types.h` contains a `mavlink_channel_t` which is used primarily in `mavlink_helpers.h` to specify the channel. The enum allows for 4 channels, MAVLINK_COMM_0 - 3. However there are functions in `mavlink_helpers` and in message definition headers (`mavlink_msg_heartbeat.h`) that use a uint8_t for channel.

Is this combination intentional? It seems like the same type (either uint8_t or mavlink_channel_t) should be used throughout the library to represent a channel.

Additionally `mavlink_channel_t` does not accurately represent the available number of comm_buffers when on linux/x86, as per the `MAVLINK_COMM_NUM_BUFFERS` define.

Personally, I would lean towards having a consistent usage of `mavlink_channel_t` throughout the library and defining it within `MAVLINK_COMM_NUM_BUFFERS` ifndef, so they are tied together and can optionally be defined externally when needed. Thoughts? Thanks!

WickedShell, 2020-04-06:
`MAVLINK_COMM_NUM_BUFFERS` can be defined before including the header which lets you define it to match your local usage/required number. (ArduPilot does this).

`mavlink_channel_t` actually should be done the sameway so that a build can provide the underlying storage type, but this hasn't been done yet.

len0rd, 2020-04-06:
Thanks for the feedback.

I'm happy to make that change and put it in a PR

len0rd, 2020-11-30:
closed via #473

closed 2020-11-30 by len0rd