ArduPilot/pymavlink #423: Struct error in sample code Mavlink_test.py

klauspeng1, 2020-06-12:
The following error message was generated when I tried to run the Mavlink_test.py code using python 3
[image]

It appears to be some problem with python struct not recognizing 's' as a bytes type for string.

For instance, the following code will result the same struct error
`import struct`
`a = struct.pack('s', "a")`
`print(a)`
[image]

TheLegendaryJedi, 2020-06-19:
Hi @klauspeng1,

You need to convert the 'a' to byte string like `a = struct.pack('s', b'a')`

peterbarker, 2020-09-23:
mavtest.py works in master now under both Python2 and Python3  - thanks for the report.

closed 2020-09-23 by peterbarker