ArduPilot/pymavlink #425: send images

pdeman, 2020-06-19:
is there any example to send images through mavlink ?
https://mavlink.io/en/services/image_transmission.html
how can I send an jpeg images ? or any open cv images ?

peterbarker, 2020-07-17:
On Fri, 19 Jun 2020, pdeman wrote:

> is there any example to send images through mavlink ?
> https://mavlink.io/en/services/image_transmission.html
> how can I send an jpeg images ? or any open cv images ?

Did you see block_xmit?

pdeman, 2020-07-17:
No, what's block_xmit ?

peterbarker, 2020-07-17:
On Thu, 16 Jul 2020, pdeman wrote:

> No, what's block_xmit ?

https://diydrones.com/profiles/blogs/demonstration-of-canberrauav-image-search-algorithm?id=705844%3ABlogPost%3A1296458&page=5

Peter

AyadAlMahturi, 2023-04-04:
@pdeman  did you end up finding an example to send images via mavlink?